bambulab/BambuStudio #6849: Slow navigation in "Prepare" 3D view after version 2.0 on macOS

dskogvold, 2025-05-07:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4.1

### Additional system information

Macbook Pro 13" 2020, M1, 16GB

### Printer

X1 Carbon

### How to reproduce

1. This problem occurs when using Bambu Studio on an external monitor with scaled resolution. It was fine before Bambu Studio version 2.0, no settings changed in Studio or macOS.
2. Open Bambu Studio and go to the "Prepare" window, on a high resolution display with scaled resolutio. I use a 27" 4K monitor scaled to 2560x1440 in macOS. Native 4K resolution works fine.
3. No need to load an actual model or project, this also happens with a new empty project.

### Actual results

Scaling monitor resolution to e.g. 2560x1440p makes Bambu Studio run slow in Prepare window, introducing lag between display updates. Rotation / navigation is hard to use. (Preview window works better). It was working fine on Bambu Studio 1.x. 

https://github.com/user-attachments/assets/84ca65c3-ea7b-415b-bddd-d0ea5f3041b4

### Expected results

Smooth 3D navigation like in Bambu Studio 1.x. It works fine if running native 4K resolution in current version too, but scaled 2560x1440 was also OK on Bambu Studio 1.x. Screen capture of same project with resolution set to 4K retina (1920x1080 but rendered in 4K like macOS does by default).

https://github.com/user-attachments/assets/72d04245-d0ed-4a67-a6b6-ece22d2454c7

### Project file & Debug log uploads

[Sampleproject.zip](https://github.com/user-attachments/files/20081906/Sampleproject.zip)

[log.zip](https://github.com/user-attachments/files/20081907/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-05-10:
Hello, we will conduct an analysis

onyxcinema, 2025-05-19:
This is similar to an issue I've reported a couple months ago. I've been sticking with 1.10.2.76 because of this. Though wish I can mess around with the H2D settings before I purchase one.