bambulab/BambuStudio #6851: Can not control Fan Speed in Bambu Studio  2.0.3.54

Nosedive4130, 2025-05-07:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Professional

### Additional system information

CPU: Intel 12900K, 64gb Ram, Nvidia 4070Ti

### Printer

Bambu Lab X1 Carbon

### How to reproduce

While running a print, 
Go to Device Tab, click on Fan inside of Control Window.

[image]

### Actual results

Pop-up opens and shows this message:

[image]

### Expected results

What should happen is that you should be able to actually control the fans.  This version and the very last version both have not had the ability to control fans from Bambu Studio on PC.

Note:  You can still control fans from the mobile app (Bambu Handy), as well as on the machine itself.

### Project file & Debug log uploads

This happens on any project that you run.  It seems the functionality has been removed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Dan-CWI, 2025-05-08:
Please bring back the fan controls in Bambu Studio. I need the manual fan controls in Bambu Studio to pre-heat the build chamber before printing ASA. The chamber pre-heats much faster with the part & aux fans running at 60%

shuizhiguo, 2025-05-09:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Nosedive4130, 2025-05-09:
Ticket No: US250509197002

Dan-CWI, 2025-05-09:
I'm on version 2.0.3.54 of Bambu Studio. Windows 10 Professional. 
In the device tab, when I click on the "Fan" button, it pops up this information box instead of displaying the fan speed controls. It is the exact same problem that Nosedive4130 is experiencing.

[image]

[image]

guywholikestobreakthings, 2025-05-10:
Same issue here.  PLEASE PLEASE PLEASE bring back this functionality.  I don't need the exhaust fan running at 100% all the time on prints.

drschlaumeier, 2025-05-11:
same issue here... X1C & version 2.0.3.54 of Bambu Studio & Win11

spaceac23, 2025-05-30:
Same, 2.0.3.54 on X1C.
Missing the fan control. I often changed the exhaust fan speed to control noise. Now it's not possible.

jmr609, 2025-06-09:
Same issue on 2.1.0.59 with X1C, getting the same "Cooling mode is suitable for printing PLA/PETG/TPU materials and filters the chamber air." dialog box as the others are reporting.

Nosedive4130, 2025-06-16:
This issue has been resolved with the update 2.1.1.52.  Thank you to all who contributed.