bambulab/BambuStudio #6853: Internal Solid Infill Pattern defaulting to Concentric even when other options are selected

vsigg, 2025-05-07:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.4.1

### Additional system information

m3 pro 

### Printer

P1S

### How to reproduce

When multi layer print is imported and internal solid infill patter is set to "Rectilinear" it defaults to concentric for the top shell layers which creates a terrible surface as the concentric lines are stacked on top of each other. It also produces a weak shell since they are not interlaced. This was not the case with older versions of Bambu Studio and is not the case with Orca slicer which tells me it has to be a bug.

### Actual results

When multi layer print is imported and internal solid infill patter is set to "Rectilinear" it defaults to concentric for the top shell layers which creates a terrible surface as the concentric lines are stacked on top of each other. It also produces a weak shell since they are not interlaced. This was not the case with older versions of Bambu Studio and is not the case with Orca slicer which tells me it has to be a bug.

[image]
[image]

### Expected results

Top shell layers need to be whatever is selected in the Global or object "internal solid infill pattern"

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-08:
@vsigg  Sorry, could you please provide the 3mf file?

QingZhangBambu, 2025-05-08:
@XunZhangBambu

vsigg, 2025-05-08:
> [@vsigg](https://github.com/vsigg) Sorry, could you please provide the 3mf file?

[ring.3mf.zip](https://github.com/user-attachments/files/20096087/ring.3mf.zip)

XunZhangBambu, 2025-05-10:
@vsigg This is because this area has been identified as a narrow region, resulting in the use of concentric filling. Using concentric filling to print narrow areas can reduce vibrations and speed up the printing.You can disable this option.

[image]