bambulab/BambuStudio #6859: No Response due to Filament Profile edeting

patrickrlr, 2025-05-08:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Monterey 12.7.6

### Printer

BambuLab X1C AMS

### How to reproduce

### Actual results

Cant close the Window of filament profile.

### Expected results

Close of window.

### Project file & Debug log uploads

[Translated Report.pdf](https://github.com/user-attachments/files/20099628/Translated.Report.pdf)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

patrickrlr, 2025-05-08:
https://github.com/user-attachments/assets/ba00418c-be11-4164-8447-391d801b6269

bambu123, 2025-05-08:
Can it be reproduced, how?

patrickrlr, 2025-05-08:
yes it can, if you edit or create more than one personalized fliament profile.