bambulab/BambuStudio #6861: Bug in Multi Device with H2D printer

hitcher73, 2025-05-08:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4

### Printer

Bambu Lab H2D

### How to reproduce

1. After enable Multi-Device, go to Mult-Device tab to add printers-
2. I have a A1 and a H2D printer.
3. I can add the A1 printer under Multi-Device, but the H2D cannot be added. It will not show up as an option. Only the A1 is visible.

[image]

[image]

### Actual results

Unable to use H2D printer under Multi-Device management.

### Expected results

Nothing.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20102300/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-09:
@hitcher73 Sorry, currently the multi-device function is not supported for H2D models.

DerGolo, 2025-08-06:
@BoBoYu-BamBu Is it on the roadmap and maybe a ETA assigned to?