bambulab/BambuStudio #6865: Online models shows up every time I open Bambu Studio, even if I disable it in preferences.

albertoMartinsen, 2025-05-08:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.4.1

### Additional system information

Mac Mini M2

### Printer

X1 Carbon + AMS

### How to reproduce

1. Open Bambu Studio and go to Settings
2. Disable "Show online staff picked models on the home page"
3. Close Bambu Studio (red dot, top left corner of the window).
4. Ensure it's totally closed (the dock icon has not the dot underneath.
5. Open Bambu Studio again clicking on the dock icon.

### Actual results

Despite it being disabled in Settings, when the app is opened, the online staff-picked content is shown up again.

[image]

### Expected results

Online staff picked content should remain disabled.

### Project file & Debug log uploads

[debug_Thu_May_08_21_00_22_94509.log.0.zip](https://github.com/user-attachments/files/20109365/debug_Thu_May_08_21_00_22_94509.log.0.zip)

A project file is not necessary or relevant to reproduce the issue.

### Checklist of files to include

- [x] Log file
- [ ] Project file

pgifford, 2025-07-17:
@MackBambu You picked up #6376, this is the same. I was disappointed to see this wasn't addressed in the beta that just came out.

MackBambu, 2025-07-29:
same #6376