bambulab/BambuStudio #6866: Unable to access online models

jamesw, 2025-05-08:
### Bambu Studio Version

2.0.3

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint

### Printer

X1C Carbon

### How to reproduce

Click online models and Makerworld.com fails to verify I am human using checkbox in cloudflare notice

### Actual results

Cloudflare fails to verify i am human

### Expected results

I expected to be able to see makerworld site

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2025-05-09:
same #6809

BobbyG333, 2025-05-12:
getting this on the newest ubuntu

closed 2025-05-20 by MackBambu