bambulab/BambuStudio #6867: Printer not loading filament since the last printer firmware update

Supanodge, 2025-05-08:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

w11

### Printer

P1S

### How to reproduce

1. I click on the history on the printer
2. select the previous print to print
3. the printer starts and just hangs and does nothing
4. bamboo studio has an error message saying Failed to get AMS mapping table; please select 'resume' to retry

### Actual results

nothing happens the printer does nothing just the fans are running and the bed and print head is keep at temperature, the printer worked fine last week i had no problems with printing.  I am printing from the printer history 

### Expected results

the printer should print 

### Project file & Debug log uploads

[Tokens.zip](https://github.com/user-attachments/files/20110342/Tokens.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

houdini91, 2025-05-09:
+1, Same thing happened to me right after i updated (i know i should have waited..)

Firmware update: 01.08.00.00

MateuszLewciki, 2025-05-09:
Same here, temperatures are good and all but there's no reaction from AMS... (I don't even get any errors in Bambu Studio).

MateuszLewciki, 2025-05-09:
I just downgraded the firmware and it's alive again!!! Check this out: https://forum.bambulab.com/t/new-firmware-doent-work-how-do-i-downgrade/47771

gaolegao-lx, 2025-05-12:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Supanodge, 2025-05-12:
Hi I downgraded my firmware and working perfectly again! 
Thanks MateuszLewciki I didn't know I could do that!