bambulab/BambuStudio #6869: Timelapse time with print by object is much too high

coat64, 2025-05-08:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

ryzen 5700G
rtx 4060

### Printer

H2D

### How to reproduce

-Make a plate with one object.
-Make a second plate and copy the object multiple times
-Set print by object
-Slice both plates
-Timelapse is set to traditional

### Actual results

Plate with one object (example) has a prepare and timelapse time of 9m28s + 6m53s
Plate with multiple objects (3 in example)  has a prepare and timelapse time of 9m28s + 1h56m

[image]

[image]

 

### Expected results

A timelapse time of 20 minutes instead of 116 minutes.

(Did not test it if the time is actually needed, this is the result the slicer gives for the H2D.
Doing the same with the X1C, no timelapse time is calculated and the print time is as expected 3 times higher for the plate with 3 objects as for the plate with the single object.)

### Project file & Debug log uploads

[testcubebyobject.zip](https://github.com/user-attachments/files/20112692/testcubebyobject.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

XunZhangBambu, 2025-05-10:
@coat64  It takes a lot time to lift to avoid collision. We will optimize this in next  version

coat64, 2025-05-10:
> [@coat64](https://github.com/coat64) It takes a lot time to lift to avoid collision. We will optimize this in next version

I don't think that is the case here. Maybe that would be the case if it was multicolor or multinozzle and the toolhead has to return to the poopchute, but that is not the case here. Also timelapse (smooth) is not possible with print by object. so also no extra traveling.

I really think it is just a bug in the representation of the used time and not in the slicing it self (again not tried it with a real print). And it seems it only applies on the H2D. On my other two machines (X1C and A1 mini) the "timelapse time" is not calculated or mentioned.

XunZhangBambu, 2025-06-03:
@coat64 This has been fixed in our latest version 02.01.00.59

closed 2025-06-03 by XunZhangBambu