bambulab/BambuStudio #6870: Can no longer change the gap of the Z seam

Soibur, 2025-05-08:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen Threadripper 3970X
RAM: 128 GB Corsair Vengeance RGB Pro 8x16GB
GPU: AMD Radeon RX 6900XT

### Printer

Bambu Lab P1S

### How to reproduce

Since the last update I couldn't find this option anymore, apparently something changed with the Z seam, there are now other things like "intelligent bias seam".

### Actual results

Since I can't change it anymore, this obviously affects the visibility of the Z seam.

### Expected results

I think that explains itself.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20112858/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Goodwine, 2025-05-11:
Duplicate of https://github.com/bambulab/BambuStudio/issues/6064

The seam gap was moved to dev mode. not sure why, but I find that decision to be completely wrong

QingZhangBambu, 2025-05-22:
Yes, I have added a process param to rewrite the filament parameters, but it has not been released yet.

closed 2025-07-21 by QingZhangBambu