bambulab/BambuStudio #6871: Allow 2 Prime Towers to be generated

hotellonely, 2025-05-09:
**Is your feature request related to a problem? Please describe.**
For H2D, the print area is splited into 3 zones, left nozzle only, common area, and right nozzle only.
If a large print (for example, Hue Forge or Light Box prints) takes up the common area and utilizes the dual nozzles, it would be beneficial to allow us to place the prime tower in Left Only or Right Only area, so that the common area can be fully utilized for printing.

However, due to the physical restriction, a single prime tower would not be good enough. Thus if the slicer can generate 2 prim towers for each nozzle, putting into each nozzle's restriction area, then it would provide lots of freedom for creation.

**Describe the solution you'd like**
Allowing placing 2 prime towers into each nozzle's restriction area.

**Describe alternatives you've considered**
Can't think of an alternative approach. At least not for H2D I think.

mctazman, 2025-05-12:
This would also work when printing 2 filaments that don't stick (well) together, like PLA/PETG.

Poyper, 2025-07-19:
Bumping this and adding to this the ability to have prime towers per object assembly in print by object mode, instead of no prime towers, or designating an object in the assembly as a prime tower to be printed first would be immensly useful.

~Pipes

marcoatgit, 2025-09-25:
Ferry useful feature request, specially too using the nozzle only area for this. I hope it will be implemented soon.

blaskotron, 2026-09-21:
Bump, this would solve several issues. 

Allow by object printing for objects with more than one color. 
By layer printing with variable layer heights between objects with prime towers. 
Variable layer heights with organic supports.