bambulab/BambuStudio #6876: Can't add a custem printer

InsaneMonkey, 2025-05-09:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: ryzen 9 3900 XT
GPU: RTX 3070
Ram: 64 GB
Display: 1440p
Storage: about 2 TB

### Printer

Ender 3 pro V1 2019

### How to reproduce

Go to "Create Printer"

The chose "Creality Ender"
And the "Ender 3 pro"

Add 0.4 nozzel size

X travel is 220 mm

Y travel is 207 mm

Z travel is 250 mm

The hit "Ok"

### Actual results

it says "The entered nozzle diameter is invalid, please re-enter"

And is does that with both custem nozzle size and the ones in the menú.

It only works when i don't use "," or "." and only work when I use a full number like "1", "2", "3", "4" or "5" and so on

Using the menú
[image]

Using custom diameter with "." first
[image]

### Expected results

It should go on to the next step, where i can chose filiment presets

### Project file & Debug log uploads

I can't get to using the slicer with my printer because of this bug, and there for there is no projekt file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

InsaneMonkey, 2025-05-09:
I don't know if it is importen, but i live en northen Europe and I use a Nordic Layout on my keyboard and windows settings

DanBao-Bambu, 2025-09-02:
The issue has been identified and is expected to be fixed in the next version