bambulab/BambuStudio #6877: Copy Paste of Text from within Bambu Studio maker world not working

tyjthomas, 2025-05-09:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 10.15

### Printer

P1S, H2D - issue is with Bambu Studio

### How to reproduce

Open any model from within makerworld. There are often links I want to follow - ie. See this GitHub repository or click here for more information. Not only do the links not work when clicked on from within Bambu Studio, but I can select the text and hit cmd + c for system wide copy and it does not copy. 

### Actual results

System cmd+c does not copy text from maker world when pulled up within Bambu Studio

### Expected results

One should be able to copy selected/highlighted text within maker world/online models on Bambu Studio - it is a universal function in every other application. If maker world is opened within browser then links and copy/paste works as expected.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GuDuJian-J-Zhang, 2025-06-11:
trying to debug

guanyun-gudujian, 2025-06-11:
trying to debug

michaelaye, 2025-12-12:
please fix ASAP.