bambulab/BambuStudio #6879: Robot verification blocking access to print history and makerworld in the PC Linux app.

Seldomscene01, 2025-05-10:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Linux Mint 22.1 x86_64 Cinnamon

### Additional system information

AMD Ryzen 5 4600G with Radeon G 
GPU: AMD ATI Radeon RX Vega 6 
GPU: AMD ATI Radeon RX 580 2048SP 
Memory: 6393MiB / 27947MiB 

### Printer

A1 and A1 mini

### How to reproduce

1 Go to home page 
2 Select "Print History"
3 Select a model
4 The following appears:
"makerworld.com
Verify you are human by completing the action below.
makerworld.com needs to review the security of your connection before proceeding."
5 Verifying makes no difference, just cycles through and step 4 reappears without end.

Same as above: 
1 Go to home page 
2  Select "Online Models"
2a A selection of models may appear on first access after restarting app
3  Select a model (A selection of models may appear on first access after restarting app
4  The following appears:
"makerworld.com
Verify you are human by completing the action below.
makerworld.com needs to review the security of your connection before proceeding."
5  Verifying makes no difference, just cycles through and step 4 reappears without end.

### Actual results

This has only just started happening. The Cloudflare "I'm Not A Robot" thing is either a bug or far too aggressive. 

It works in windows app (I think) but to check, I have to leave this page and lose it switching OSs. I don't want to use Windows as I only use it for 1 piece of hardware I have and I use CoralDraw.  All my stuff is set up for L:inux

### Expected results

Works on models I look at on MakerWorld Web page and downloaded from there. 

### Project file & Debug log uploads

[Log.zip](https://github.com/user-attachments/files/20132039/Log.zip)
Problem in app no matter which file as any file can't be accessed.

### Checklist of files to include

- [x] Log file
- [ ] Project file

GabeNedden, 2025-05-11:
I'm also experiencing this

Ronny7177, 2025-05-11:
Same 4 me

MackBambu, 2025-05-13:
same https://github.com/bambulab/BambuStudio/issues/6809

closed 2025-05-20 by MackBambu