bambulab/BambuStudio #6880: AMS 2 Pro does not load filament for flow rate calibration

FrostIronMedia, 2025-05-10:
### Bambu Studio Version

01.09.00.01

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1 Carbon

### How to reproduce

1. Go to flow rate calibration
2. Select AMS 2 Pro 
3. Select B1-B4 filament slots
4. Attempt to calibrate

### Actual results

The printer will not load the filament to calibrate. It will still attempt to calibrate but with no loaded filament. If filament is already loaded, even in the same slot, it will pull the filament back but not load again and will continue. AMS 2 works fine, but AMS 2 pro does not when both are connected. 

### Expected results

I should be able to calibrate the flow with AMS 2 Pro as I can with AMS 2.

### Project file & Debug log uploads

[debug_Fri_May_09_21_57_34_35356.log.zip](https://github.com/user-attachments/files/20137738/debug_Fri_May_09_21_57_34_35356.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-12:
Sorry, could you please provide the printer log?
 The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/x1/troubleshooting/export-failed-print-log-to-sd-card

changbaishen, 2025-05-19:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

lucgei231, 2025-05-20:
I have this same issue, but it occurs with AMS original model.

I have an X1C with 3x AMS and 1x AMS Pro. AMS units in the chain in position 2,3,4 won't feed filament for automatic calibration.

ageitgey, 2025-05-20:
I have the same issue where my X1C with 4 AMS units won't feed filament when running automatic flow raw calibration. Since no filament is fed from the AMS, the calibration runs with nothing in the nozzle and then fails.

I have 3 AMS and 1 AMS 2 Pro attached. The issue happens with any AMS unit I've tested, so the issue is not limited to AMS 2 Pro. This issue happens with filament slots B1-B4, C1-C4, D1-D4 which are all original AMS units. I did not test filament in slots A1-A4.

It seems there is another report here as well: https://github.com/bambulab/BambuStudio/issues/6808#issuecomment-2860851165

FrostIronMedia, 2025-05-20:
> I have the same issue where my X1C with 4 AMS units won't feed filament when running automatic flow raw calibration. Since no filament is fed from the AMS, the calibration runs with nothing in the nozzle and then fails.
> 
> I have 3 AMS and 1 AMS 2 Pro attached. The issue happens with any AMS unit I've tested, so the issue is not limited to AMS 2 Pro. This issue happens with filament slots B1-B4, C1-C4, D1-D4 which are all original AMS units. I did not test filament in slots A1-A4.
> 
> It seems there is another report here as well: [#6808 (comment)](https://github.com/bambulab/BambuStudio/issues/6808#issuecomment-2860851165)

So my issue is similar but different. My AMS works fine. Will calibrate per usual. But my second unit my ams2 pro will not and will do the same as yours which is not load the filament but try to calibrate.

It will calibrate when its the only unit connected however so im assuming it has to do with the multiple units because thats where we both have a similar issue. 

Ive sent a ticket with my logs twice but they keep asking for them. Not sure what the logs will show though other than a failed calibration but it will get fixed once enough ams 2 pros go out 🤣

ageitgey, 2025-05-20:
> im assuming it has to do with the multiple units because thats where we both have a similar issue.

Yeah, I bet that's the root cause.

For what it's worth, I'm using Bambu Studio on Mac. So it doesn't seem to be a platform-specific issue.

> Ive sent a ticket with my logs twice but they keep asking for them. 

When I've reported legit bugs in the past, they were silent and then suddenly responded to confirm them as fixed in the next release hours before the new release dropped. So who knows?

shiftylilbastrd, 2025-05-20:
My original AMS is doing the same thing so this isn't exclusive to AMS2

amebos, 2025-05-21:
I have the exact same issue after installing a 0.6 mm nozzle.  First flow calibration indicated a mismatch, even though I set 0.6 on the machine and in the calibration settings.  After a reboot, it runs the calibration but does not load the filaments as described above at all.  If I pre-load the filament I want to calibrate, the AMS unloads it during the calibration, then does the calibration steps without filament.  

Has anyone had luck unplugging all but one AMS as a workaround for the time being?

BoBoYu-BamBu, 2025-05-22:
Please upload log through Handy APP, and give us your ticket number

Upload log files through Bambu Handy
Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
To upload files, go to Me → Support Tickets → Choose the ticket for log upload and proceed with the operation.
app_support_ticket
[image]

natemccurdy, 2025-08-27:
I'm having the same problem on my P1S. The AMS doesn't even attempt to load the chosen filament when doing a Flow Rate Calibration, and the print always throws the following error:

[image]

### Ticket

I've opened a ticket via Bambu Handy: `US250827731001`

### Versions

* Bambu Studio: `2.2.1.58`
* Bambu Lab P1S: `01.09.00.00`
* AMS 2 Pro: `02.00.19.68`

### Workaround

I've found a workaround that helps some of the time.

If I preload the filament with the Bambu Handy app (Devices -> Filament -> View all -> click a filament -> click load) **then** I start a calibration with the chosen filament, it'll work most of the time. Though I do still get the pop up about filament running out; however, I can just click "Filament loaded, resume", and the calibration print will eventually complete.