bambulab/BambuStudio #6888: colori e filamenti dopo invio anteprima

LaserMancas3D, 2025-05-11:
### Bambu Studio Version

ultima 

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Additional system information

buongiorno , il problema e' che una volta generato e modificato un oggetto di vari colori e anche dopo averli uniti ,nella descrizione dell oggetto a destra sul programma i colori ci sono e anche a video si vedono

[image]
[image]

 ,ma al momento dello slice sparisce uno o piu colori ed il filamento corrispondente al colore nella finestra slice filamenti  , anche nella finestra di stampa il colore sparisce uno o piu colori . lo fa a volte e a volte no me eseguendo i medesimi comandi e modifiche all oggetto anche di vari file differenti.  lo fa sia connesso che disconnesso alla A1 con Ams . 

### Printer

bambulab A1

### How to reproduce

[image]
[image]

### Actual results

[image]
[image]

### Expected results

[image]
[image]

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-05-12:
@LaserMancas3D, Thank you for the feedback.  from the screenshots, we see a connection failure. The studio can not connect to the printer? 
[image]
Could you please reproduce the problem, and offer some logs.https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

XinZhangBBL, 2025-07-01:
Halo, we have solved the problem at latest release. please check https://github.com/bambulab/BambuStudio/tags