bambulab/BambuStudio #6889: Step-import gives messy result

coat64, 2025-05-11:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

Amd 5700G
RTX 4060

### Printer

H2d

### How to reproduce

Open step-file of curved object.

### Actual results

In the prepare tab you can already see the import is a little messy. Added a picture of the same model exported as STP and STL.

[image]

When sliced the slice is messy.

[image]

The exported STP is not flawed, when importing in other program it is properly build up:

[image]

### Expected results

Import step-files used to be better.

### Project file & Debug log uploads

[stp_messy.zip](https://github.com/user-attachments/files/20148552/stp_messy.zip)

The stp-file is included

### Checklist of files to include

- [ ] Log file
- [x] Project file

coat64, 2025-05-11:
I see now that there is a option in the preference for step-files:

[image]

I think it is unchecked in the latest release?

Anyway when setting the both parameters to its minimum you get a good result. But the default setting is really to low.

MackBambu, 2025-05-12:
> I see now that there is a option in the preference for step-files:
> 
> [image]

Yes, enabling this dialog and adjusting the options can effectively improve the mesh quality.

> I think it is unchecked in the latest release?
> 
You may have previously selected the "Don't show again" option when the dialog was shown, so it has been remembered.

> Anyway when setting the both parameters to its minimum you get a good result. But the default setting is really to low.

This default value is unlikely to suit all models. When your model is very complex, setting this value too small may cause extremely slow loading. However, if the model has many curved surfaces but a simple overall structure, it may be necessary to use a smaller value to achieve higher quality.

coat64, 2025-05-12:
> You may have previously selected the "Don't show again" option when the dialog was shown, so it has been remembered.

You are probably correct about this, thanks for the explaining, appreciated.

closed 2025-05-13 by MackBambu