bambulab/BambuStudio #6893: Scarf seam position wrong.

dkbay, 2025-05-11:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macos 15.4.1

### Printer

Bambulab X1C

### How to reproduce

1. Have an object with a round part that also has a sharp corner in one area.
2. Have conditional scarf seam enabled.
3. Bambu studio will place the end of the scarf seam aligned with the normal seam instead of placing the start of the seam with the normal seam.

See image:
[image]

### Actual results

Seam is offest by its length from the normal seam causing it to look like it is placed in the middle of nowhere.
Also the seam "bends" just before the normal seam is placed on the shape corner

[image]

### Expected results

I'd expect it to align the seams like in prusaslicer where the actual seam dot is aligned the same with both seam types.

[image]

### Project file & Debug log uploads

I can't.
This is a general problem should be easy to reproduce without it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2025-05-22:
You want to place the end position of the loop at the original point. However, the start of the loop may become an obvious point. This point is the seam position. 

[image]

dkbay, 2025-05-22:
@QingZhangBambu If that is the case I would expect the white seam dots to be placed there.
Generally I would expect the seam dots to be placed where the seam is the most visible so I can have an idea what my part will look like.

QingZhangBambu, 2025-05-22:
@dkbay Yes, it should be. However, the slicer is unable to know the actual print result.

dkbay, 2025-05-26:
You and i can know the result though.
So if anything maybe we could get to pick how it aligns? 

It's just really annoying as it is right now I have scarfed seams that start/end out of line from everything else.