bambulab/BambuStudio #6896: Bambu studio crashing when slicing

uaag, 2025-05-12:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Ryzen 9 5950x
RAM: 32 GB
GPU: 4080 Super

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Just trying to slice a model and the software crashes

### Actual results

Bambu Studio Crashes

### Expected results

Bambu studio crashes

### Project file & Debug log uploads

[MagLevMoonAndBase.zip](https://github.com/user-attachments/files/20152906/MagLevMoonAndBase.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-12:
@uaag  Sorry, could you please provide the log and DMP files?  The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

uaag, 2025-05-12:
[crash_Sun_May_11_19_40_11_0_done.log](https://github.com/user-attachments/files/20154774/crash_Sun_May_11_19_40_11_0_done.log)

I can not uplaod the dmp file, github is giving me an error, but here is the log.

Thanks

BoBoYu-BamBu, 2025-05-12:
@uaag Could you please send the DMP file to my email?My email address is:yuting.huang@bambulab.com；This log file doesn't show where the error occurred.

uaag, 2025-05-12:
hello @BoBoYu-BamBu its a bit heavy, around 2.5 GB, so Ill upload it to my google drive and share the link to it. Is that okay? Ill do it later today as I am at work right now.

Have a nice day!

Thanks!

BoBoYu-BamBu, 2025-05-13:
@uaag OK, thank you very much.

uaag, 2025-05-14:
hello @BoBoYu-BamBu sorry for the delay, you can get the dmp file [here].(https://drive.google.com/file/d/13eQcOUtMpuoWS9oWKmcGN9sJa_iMvFhf/view?usp=sharing)

Thanks!

QingZhangBambu, 2025-05-22:
@uaag Could you try it to see if it works
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

@XunZhangBambu it crash on arachne, could you check it?

[image]