bambulab/BambuStudio #6897: Unable to connect to Printer in LAN-Mode

g4njawizard, 2025-05-12:
### Bambu Studio Version

2.00.03.54

### Where is the application from?

Bambu Lab github releases

### OS version

Debian

### Printer

X1C

### How to reproduce

Download the Appimage, or use the flathub version.
Set your printer to LAN-Mode only.
Connect to the same network (WLAN or LAN or whatever..)
Try to connect to the printer via "Bind with Access Code"

### Actual results

Nothing happens. (No errror message)

### Expected results

Connection to the printer over LAN-Mode

### Project file & Debug log uploads

[BambuStudioLog.zip](https://github.com/user-attachments/files/20154465/BambuStudioLog.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

g4njawizard, 2025-05-12:
I am able to use a Home Assistant Plugin and access the printer using the IP, access code and the serial number. Why isn't that possible with BambuStudio??

hotellonely, 2025-05-12:
> I am able to use a Home Assistant Plugin and access the printer using the IP, access code and the serial number. Why isn't that possible with BambuStudio??

it's only showing up for the "first" instance of bambu studio running at the same time, in lan only mode.

g4njawizard, 2025-05-12:
But homeassistant is no bambustudio. It just requests data, but doesnt act as a slicer.
So this means I have to decide what I want to connect, but I cannot have both?
That doesnt make sense to me.
This means 2 me that If I run a company and have 5 of those printers, I can only access them from 1 client at a time? Or do I need to recreate the access code so both clients have "their own" access code?

hotellonely, 2025-05-12:
each computer can have only 1 bambu slicer window at a time, but you can have multiple computers.

g4njawizard, 2025-05-12:
Please clarify, multiple computers per printer, or one printer for multiple computers?
This is important to understand, because currently I try to find a way to achieve a certain level of privacy.
Having Home Assistant to close a gap where the Bambu App is not usable in LAN-Mode, is my choice for a safe remote observation.
But I want to initially start a print with bambu studio on my client without using the cloud, but as I understand, I cannot have both?

hotellonely, 2025-05-12:
> Please clarify, multiple computers per printer, or one printer for multiple computers? This is important to understand, because currently I try to find a way to achieve a certain level of privacy. [...]

I mean you can have multiple computers to connect at the same printer. 
What you're saying is doable and people are doing that right now.
You don't have to start the print via the cloud when using Bambu Studio. It works in Lan Only mode. The problem is that your Bambu Studio don't see your printer in your network. And the reason of that is pretty complicated. All I can say is that Bambu Studio works when:
1. your computer and your printer is under the same subnet, not crossing a vpn or something.
2. all the necessary ports are accessible.
3. you only have 1 bambu studio instance running on that computer. it doesn't matter if you have other computers running bambu studio or not.

If you still have issues, then it's 99.99% your network settings.

g4njawizard, 2025-05-12:
Then this has to be a bug, since I am like initially posted connected on the same subnet (in my case I use WLAN, since this printer has no LAN port..)

I dont think my client with bambustudio is having network issue's. I only use default iptables (linux fw) rules which dont drop broadcasts of some kind.

`
Chain INPUT (policy DROP)
target     prot opt source               destination         
ufw-before-logging-input  all  --  anywhere             anywhere            
ufw-before-input  all  --  anywhere             anywhere            
ufw-after-input  all  --  anywhere             anywhere            
ufw-after-logging-input  all  --  anywhere             anywhere            
ufw-reject-input  all  --  anywhere             anywhere            
ufw-track-input  all  --  anywhere             anywhere            

Chain FORWARD (policy DROP)
target     prot opt source               destination         
ufw-before-logging-forward  all  --  anywhere             anywhere            
ufw-before-forward  all  --  anywhere             anywhere            
ufw-after-forward  all  --  anywhere             anywhere            
ufw-after-logging-forward  all  --  anywhere             anywhere            
ufw-reject-forward  all  --  anywhere             anywhere            
ufw-track-forward  all  --  anywhere             anywhere            

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
ufw-before-logging-output  all  --  anywhere             anywhere            
ufw-before-output  all  --  anywhere             anywhere            
ufw-after-output  all  --  anywhere             anywhere            
ufw-after-logging-output  all  --  anywhere             anywhere            
ufw-reject-output  all  --  anywhere             anywhere            
ufw-track-output  all  --  anywhere             anywhere            

Chain ufw-after-forward (1 references)
target     prot opt source               destination         

Chain ufw-after-input (1 references)
target     prot opt source               destination         
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:netbios-ns
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:netbios-dgm
ufw-skip-to-policy-input  tcp  --  anywhere             anywhere             tcp dpt:netbios-ssn
ufw-skip-to-policy-input  tcp  --  anywhere             anywhere             tcp dpt:microsoft-ds
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:bootps
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:bootpc
ufw-skip-to-policy-input  all  --  anywhere             anywhere             ADDRTYPE match dst-type BROADCAST

Chain ufw-after-logging-forward (1 references)
target     prot opt source               destination         
LOG        all  --  anywhere             anywhere             limit: avg 3/min burst 10 LOG level warn prefix "[UFW BLOCK] "

Chain ufw-after-logging-input (1 references)
target     prot opt source               destination         
LOG        all  --  anywhere             anywhere             limit: avg 3/min burst 10 LOG level warn prefix "[UFW BLOCK] "

Chain ufw-after-logging-output (1 references)
target     prot opt source               destination         

Chain ufw-after-output (1 references)
target     prot opt source               destination         

Chain ufw-before-forward (1 references)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED
ACCEPT     icmp --  anywhere             anywhere             icmp destination-unreachable
ACCEPT     icmp --  anywhere             anywhere             icmp time-exceeded
ACCEPT     icmp --  anywhere             anywhere             icmp parameter-problem
ACCEPT     icmp --  anywhere             anywhere             icmp echo-request
ufw-user-forward  all  --  anywhere             anywhere            

Chain ufw-before-input (1 references)
target     prot opt source               destination         
[... 81 lines cut ...]

Chain ufw-user-input (1 references)
target     prot opt source               destination         

Chain ufw-user-limit (0 references)
target     prot opt source               destination         
LOG        all  --  anywhere             anywhere             limit: avg 3/min burst 5 LOG level warn prefix "[UFW LIMIT BLOCK] "
REJECT     all  --  anywhere             anywhere             reject-with icmp-port-unreachable

Chain ufw-user-limit-accept (0 references)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             anywhere            

Chain ufw-user-logging-forward (0 references)
target     prot opt source               destination         

Chain ufw-user-logging-input (0 references)
target     prot opt source               destination         

Chain ufw-user-logging-output (0 references)
target     prot opt source               destination         

Chain ufw-user-output (1 references)
target     prot opt source               destination   
`

logikal, 2025-05-12:
I'm having a similar issue with my H2D.

I haven't had the experience that @hotellonely has had: I can have multiple instances of BambuStudio or OrcaSlicer on the same computer all see my X1C in LAN mode.

@g4njawizard: are you able to see the broadcasts from the printer? 

```
export PRINTER_IP="nn.nn.nn.nn"
sudo ngrep -d any -Wbyline port 1990 or port 2021 or port 8883 or port 900 or port 322 or port 6000 or host $PRINTER_IP or host 239.255.255.250
```

You should see some ssdp traffic at least - this is how studio discovers LAN printers, afaict.

g4njawizard, 2025-05-15:
There must be a relation between the network driver in bambu studio and the printer.
I tried older Bambu Studio versions and all wanted to install updates when launching the application.
I also tried flatpak and appimage versions.. no difference.

Either Bambu's X1C firmware has some issues or ur network plugin. I think it might be related to your network plugin.

Utility9298, 2025-05-16:
I am having the same issue on windows and X1C with firmware v01.09.00.01
BambuStudio v2.0.3.54

Seems like it is trying to connect and then the connect-window closes but printer is not shown in BambuStudio.

bromic1, 2025-05-20:
I have the same issue with A1 and firmware v01.03.30.01.
BambuStudio v2.0.3.54
This is a nogo. I purchased the printer, got it and i was searching for a failure for 1.5h until I searched github issues and saw it is a software bug!

Utility9298, 2025-06-20:
Did just upgrade to Firmware Version 01.09.01.00 on X1C. 
Still not able to connect via IP adress across diferent subnet (I double checked, traffic is not blocked by firewall)

harry-bbl, 2025-07-31:
@g4njawizard Please update to version 2.2 beta and verify again. If the issue persists, please attach the log file to this issue. I will follow up on this issue continuously.

g4njawizard, 2025-07-31:
Seems to work. I could connect to my printer and calibrate via bambu studio.
Thanks for taking care @harry-bbl .