bambulab/BambuStudio #6898: URGENT: H2D Left Nozzle Auto Calibration destructs nozzle stopper in newest beta firmware.

hotellonely, 2025-05-12:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

UNRELATED.

### Printer

H2D

### How to reproduce

1. Go to Calibration page after connected to H2D, with newest beta firmware.
2. Apply automatic flow calibration on a filament that is on the left nozzle.
3. See this happen in the after print cleaning process:

### Actual results

[image]

### Expected results

Fix this behavior and prevent damaging the nozzle&nozzle stopper mechanism.

### Project file & Debug log uploads

No you simply start the flow dynamics calibration from the software.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hotellonely, 2025-05-12:
Submitting this in courtesy of @2B.Mainprints from Bambu user forum. 
https://forum.bambulab.com/t/h2d-beschadigt-sich-selbst/168438
The original post is in German so it's not reaching enough population yet. But as the severity of the issue is pretty high I would suggest a fix ASAP.

BoBoYu-BamBu, 2025-05-12:
@hotellonely Thank you for your feedback. We are currently confirming this issue.

hotellonely, 2025-05-13:
> **Please upload log through Handy APP, and give us your ticket number**
> 
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions. When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration. To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation. [image]

I'm submitting this in courtesy of @2B.Mainprints from Bambu user forum.
https://forum.bambulab.com/t/h2d-beschadigt-sich-selbst/168438
The original post is in German so it's not reaching enough population yet. But as the severity of the issue **I am obviously not going to destroy my own printer just to test**. You can test it on your end easily if you  read the original post on your forums, with the link above.

2B-MainPrints, 2025-05-13:
Hi - here ist 2B.Mainprints - i submitted the Issue in theoffical Forum from Bambulab.
I also created a ticket with the logfiles and a short video of the error 

here is the Ticket Nr.: US250511538001