bambulab/BambuStudio #6899: won't sync correctly anymore with Bambu AMS Lite.

KB46, 2025-05-12:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

widows 11

### Printer

Bambu A1 with AMS lite

### How to reproduce

import stl or .3mf, select printer, hit sync with ams button, overwriting button or mapping, Sync now button, success popup- OK tab. Changed to wrong filament on AMS and won't change at all from that first time. Used to work using overwriting button.

### Actual results

Changed first time to wrong filaments and only 3 spools mapped wrong even with 1 spool in ams.

### Expected results

All filament types should be listed in ams

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20166544/log.zip)

[ROSE_STANDARD_GRANDMA.zip](https://github.com/user-attachments/files/20166608/ROSE_STANDARD_GRANDMA.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-13:
@KB46 Thank you for your feedback，Could you please provide a screenshot or a video recording to describe your issue? I'm not quite understanding it
Thanks！

Haidiye00, 2025-08-04:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest version and confirm again if there are any issues. If there are any problems, please submit a GIF. Thank you