bambulab/BambuStudio #6900: Wont recognize external spool & AMS Lite at same time. Version 2.1 had no issues.

Video228, 2025-05-12:
### Bambu Studio Version

02.00.02.57 & 02.00.03.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu lab A1 & AMS Lite

### How to reproduce

issue with the last 2 updates to Banbu studio not allowing you to shut off the AMS before sending the print to the A1? Version v02.00.01.50  had an advanced option that allowed you to toggle of “use ams” and then you could choose the external spool, which was fantastic. But Bambu studio version v02.00.02.57 did not allow you to pick external or AMS, and everything was greyed out. And most recently version v02.00.03.54 was same thing! I posted this on the help group of bambu labs, and some people recommended unplugging the ams lite. I work remotely, and do not have that option. I show spool 5 as external, and A! has no issue recognisong is as tpu. I am posting a few pics that show what Google ai says when asking “how to disable ams in banbu studio” just to show it responds as it did exist. But the link is to things from 2024. Am i the only one asking this question? Or worse. The only one that this is happening to? Am i stuck on a bambu studio verion that was a year old to make things complicated? I followed the instructions online about version 2.2 and tried to choose the external spool, and get a message "only ams slots with same material can be selected". I did not want nor choose ams slots. please see pics labeled v2.1, 2.2 & 2.3 2.2 and 2.3 act exactly the same. Bambu studio was broken in your last 2 updates.

[image]
[image]
[image]
[image]
[image]
[image]

.

### Actual results

nothing happens. it freezes me from printing anything.

### Expected results

I should be able to pick "ONLY" the external spool, but it stays greyed out.

### Project file & Debug log uploads

It does it on all files.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-13:
@Video228 Sorry, we are currently working on resolving this issue. You can first set the external consumables to TPU on the device page, then manually select the external consumables on the print page when sending the print job, and it should be able to be sent and printed.

XinZhangBBL, 2025-07-04:
The problem has been solved, please check the latest release