bambulab/BambuStudio #6914: H2D Beta Firmware new behavior causing nozzle sticking into purge strip and stick purged filament onto printed part.

hotellonely, 2025-05-13:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab github releases

### OS version

Unrelated

### Additional system information

Unrelated

### Printer

Bambulab H2D

### How to reproduce

1. Using High Flow 0.4mm nozzle.
2. Print something in PETG HF.
3. At layer 3 and layer 10, nozzle would go to purge strip and dip into it.
4. Now there're purged old filament sticking on the nozzle.
5. Sticked filament would eventually end on part, destroying wall quality.

### Actual results

[image]
For this print, the first layer and concecutive layers were very clean, until the nozzle brought some purged poop back to the wall. And it's sticking on the next layer right where the travel ends.

### Expected results

Before the release of official firmware, you should either:

1. Disable this feature (I assume it's some sort of clump detection or pressure advance calibration?)
or:
2. Make the first purge line shorter, move it a bit, let the new purge move to somewhere that would not dip into old purge line.

I noticed the same behavior to happen on both PETG and PLA, but because bambu nozzle doesn't have anti-sticking coating, it's very obvious on PETG + high flow nozzle.

I don't think Bambu would need my log because it's a very consistent behavior, all you need is just a high flow nozzle and the petg hf filament.

### Project file & Debug log uploads

1. Create a new project, maybe just a benchy or calibration cube, anything.
2. select filament and nozzle type to petg hf and high flow nozzle.
3. print and watch how the nozzle dips into the purge strip, while it might not bring some filament EVERYTIME, it's dipping into the strip and causing a high chance of sticking some filament back.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hotellonely, 2025-05-13:
The purge strip:

[image]

You can see the "second purge" started right on top of the tail of the first purge strip.

[image]

It seems it's either the "second purge" or "third purge" that is dipping into previous purge, causing issues in the future printing.

Please fix this simple bug asap :(

magyoshaa, 2025-05-13:
i have same behavior with standard 0.4mm nozzle at similar layers

hotellonely, 2025-05-13:
> i have same behavior with standard 0.4mm nozzle at similar layers . i think it was doing it also pre beta frimware so maybe not related to that

I think it was not doing that pre beta firmware? it's a similar behavior as the a1 series for nozzle clumping detection, but executed poorly.

magyoshaa, 2025-05-14:
> > i have same behavior with standard 0.4mm nozzle at similar layers . i think it was doing it also pre beta frimware so maybe not related to that
> 
> I think it was not doing that pre beta firmware? it's a similar behavior as the a1 series for nozzle clumping detection, but executed poorly.

rolling back my firmware to test as its happens every print so should notice it quickly

magyoshaa, 2025-05-14:
> > i have same behavior with standard 0.4mm nozzle at similar layers . i think it was doing it also pre beta frimware so maybe not related to that
> 
> I think it was not doing that pre beta firmware? it's a similar behavior as the a1 series for nozzle clumping detection, but executed poorly.

you are correct.. its not doing it on rolled back firmware.. only on beta.