bambulab/BambuStudio #6915: Circular path is 0.68% smaller than square path in G-code.

warp929, 2025-05-13:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 7 3700X 8-Core Processor
32GB RAM

### Printer

H2D

### How to reproduce

1. Import a model that is a cube (27.940mm for x and y, 13.970 for z)  with a 19.304mm diameter cylinder on top. Cylinder height is 3.810mm
2. Generate 3mf file with 0.20mm layer height
3. Unzip 3mf file
4. Open g-code file
5. Isolate a layer in the middle the cylinder to only show that arc-path or segments that make up a circle.
6. Determine the radius of the circle based on arc information or linear segments. Use Excel or Gemini (gives same answer)
7. Isolate a layer in the middle of the cube to only show the path of the square.
8. Determine the maximum and minimum for the x and y coordinates.
9. Now calculate the g-code steps/mm for the square and for the circle.
10. Compare the circle steps/mm to the square steps/mm. 

### Actual results

The steps/mm for the circle path is consistently 0.68% smaller than the square. 

### Expected results

The steps/mm for the circle path should be the same as for the square path. 

It doesn't matter if you select "Arc fitting" or not. It does not matter if the Resolution is course or fine. The results are always the same. The circle is too small. 

[image]

### Project file & Debug log uploads

[Block with cylinder for github.zip](https://github.com/user-attachments/files/20192966/Block.with.cylinder.for.github.zip)

There are no Errors or Log files for this particular issue.

### Checklist of files to include

- [x] Log file
- [x] Project file

warp929, 2025-05-14:
> saw your post in forum, gotta give you a bump up here :)

Thank you.

warp929, 2025-05-15:
I have reanalyzed all my data to be sure that I didn't make an error.  I output new files from Solidworks and Fusion360. I used various file formats: STL, 3MF, and STEP. 

I then validated my calculations using Gemini AI Pro and ChatGPT.  No errors were found.

The problem appears consistently through all slicer programs and I can't figure out where the error is.

[image]

[Slicer Analysis.xlsx](https://github.com/user-attachments/files/20231269/Slicer.Analysis.xlsx)

QingZhangBambu, 2025-06-03:
i check the data of your 3mf, the circle d is 204.182- 9.9998 => 194.182=>19.4 mm, so the r is 9.7091
gcode path as show in the pic. 
[image]

the width of the outer wall is 0.42, so the path radius is about 9.4991
[image]

QingZhangBambu, 2025-06-03:
according to the data, the path of gcode is right.

warp929, 2025-06-03:
I respectfully disagree. I believe the path is a bit more complicated than simply accounting for the line width.  The important thing is that is does not matter if you are making relative measurement between the square and the circle. 

I compare the g-code radius to the true radius (9.652mm) and get an independent conversion value. I then compare the g-code maximum and minimum X values to get a width of the square. That g-code width is then compared to the true width of the square and you get another independent conversion value.

Now compare the two independent conversion values. They should be the same but the circle conversion value is 0.68% smaller. Why would that be? In an actual printed part, this error is measurable and significant when mating to other parts.

I'm attaching my Bambu Studio project file and my Excel file.

[image]

[Excel analysis.xlsx](https://github.com/user-attachments/files/20564210/Excel.analysis.xlsx)

[Block with cylinder for github as Bambu Studio Project.zip](https://github.com/user-attachments/files/20564229/Block.with.cylinder.for.github.as.Bambu.Studio.Project.zip)

QingZhangBambu, 2025-06-03:
Different segments of curves may result in various diameters of the fitted arcs.

QingZhangBambu, 2025-06-06:
so you think the correct radius to be calculated should be 9.65 instead of not 9.448?

QingZhangBambu, 2025-06-06:
thanks for reply, i get your point. 
The difference in this error should cause by offset algorithm used in the calculations.

[image]

[image]

warp929, 2025-06-06:
My calculation method is independent of the shrink factor because the shrink factor would be applied to the circular path and the square path equally. I have tested with shrink factor at 100% and 99.4%. The error between the square and the circle are the same.

Let me try with a separate example to make my point:
CAD outputs a 20mm x 20mm square. It also outputs a 20mm diameter circle. This is at very high resolution (thousands of triangles) into an STL file. I used Solidworks and Fusion 360 (pay version). I tried STL, STEP, and 3MF with similar results.

Bambu Studio calculates the path of the square to be  19.699mm x 19.699mm.
The relationship between the CAD and the Gcode is: 
Gcode/CAD = 19.699/20 = 98.497%

Bambu Studio calculates the diameter of the circle to be 19.563mm
The relationship between the CAD and the Gcode is: 
Gcode/CAD =19.563/20 = 97.815%

The difference between the square and the circle is:
98.497-97.815 = 0.68%

The .68% error makes circles too small compared to their square. This makes it impossible to generator a proper part because the relationship is incorrect. Anything that is done to make the circle bigger will cause the square to be become too big.

.68% is bigger than the shrink factor for ABS so I find that a significant error in gcode path calculation.  I believe Bambu Studio is based on Slic3r which also has a similar error.

QingZhangBambu, 2025-06-06:
This is a complex issue.

The path generated by GCode is the centerline path of extrusion, and the dimensions of the final contour should consider the extrusion width added to this centerline path.

 A circular contour can be slightly smaller than a square diameter contour. This is because the calculation of the centerline is based on the original contour shrinking inward by half a line width. As illustrated in the previous discussion, this might lead to a slightly larger dimension. 
However, for a circle, the calculation discrepancy in value should not reach 0.1mm.
As per the data you provided, with an error of 0.002mm, this level of discrepancy is acceptable.

Moreover, based on the principle behind this error generation, even if the diameter of the circle increases, this error should not become larger.

If you set a 20mm square, but the slicer result is 19.699, please provide the 3MF file. I will check the original contour data and the G-code path data, as it seems that the size of the square is slightly larger.