bambulab/BambuStudio #6916: Bambu Studio 2.0.3.54 crashing during slicing

domenick, 2025-05-13:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10Pro

### Additional system information

CPU 12th Gen Intel (R) Core (TM) i9-12950HX, 64 GB RAM, NVIDIA RTX A4500 Laptop GPU (15GB) UHD (128MB)

### Printer

Bambu X1C

### How to reproduce

Seems to be arbitrary. Doesn't always happy, even on the same file. It happens more often on multicolor files.

### Actual results

program just closes and when I reopen it it says do you want to reinstall project, I say yes and then I will slice and it works. If right after that I make a change to the slicing parameters it may or may not close again.

### Expected results

It shouldn't close.

### Project file & Debug log uploads

[Doberman_Pose_03.stl.zip](https://github.com/user-attachments/files/20195259/Doberman_Pose_03.stl.zip)

[crash_Tue_May_13_14_27_52_0_done.log](https://github.com/user-attachments/files/20195268/crash_Tue_May_13_14_27_52_0_done.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-14:
@domenick Sorry, could you please provide the  DMP files?

domenick, 2025-05-14:
This was today

[debug_Wed_May_14_07_59_34_37976.log.zip](https://github.com/user-attachments/files/20207020/debug_Wed_May_14_07_59_34_37976.log.zip)

and this is yesterday

[bambu-studio.exe.11412.zip](https://github.com/user-attachments/files/20207056/bambu-studio.exe.11412.zip)

[debug_Tue_May_13_14_31_30_11412.log.zip](https://github.com/user-attachments/files/20207083/debug_Tue_May_13_14_31_30_11412.log.zip)

domenick, 2025-05-23:
@BoBoYu-BamBu I have not heard from anyone in over a week. Any info?

domenick, 2025-05-24:
@Haidiye00 are you asking me to do something?

domenick, 2025-05-26:
@Haidiye00 I tried it out and it hasn't crashed yet but I will try to get it to crash. To be honest I have not had any further crashes. Perhaps it was the Doberman file I was using. I will try that and run it a few times. I'll let you know.

domenick, 2025-05-31:
@Haidiye00 I continued to try and crash the temp package and could not do it. I hope this helps. I did have another crash on my Studio using a different file, but it had been a long time since the last time. I attached the dmp file.

[bambu-studio.exe.7448.zip](https://github.com/user-attachments/files/20534507/bambu-studio.exe.7448.zip)

closed 2025-06-03 by Haidiye00

guanyun-gudujian, 2025-07-14:
@domenick 
Hi, pls try this package and  help to check if it can slice without crashing
thanks
https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16257448317/artifacts/3523837507