bambulab/BambuStudio #6917: Printer not showing up in Prepare in BambuStudio on Mac

jshort-indy, 2025-05-13:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.4.1

### Additional system information

Apple M3, 16 GB

### Printer

Bambu Lab P1S

### How to reproduce

My printer does not show up in the Prepare tab.  I have a P1S showing but it is not connecting to my P1S and attached AMS.  The printer is showing up in the Device tab and the AMS information is updating.

### Actual results

See comment above.

### Expected results

See comment above

### Project file & Debug log uploads

See comment above.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-14:
@jshort-indy  Thank you for your feedback. Due to network issues, the printer is still in the connection process and thus the AMS is not displayed. Could you please try again now? If it still doesn't work, could you provide the log file? Thank you.

BoBoYu-BamBu, 2025-06-06:
@jshort-indy  Have you solved your problem yet?