bambulab/BambuStudio #6918: Remove outline when moving objects (or add setting to disable it)

barneyvaughan, 2025-05-13:
**Is your feature request related to a problem? Please describe.**
In previous versions the movement tool selected an object without a white outline which allowed for more precise control of the object you were manipulating. Now the white border is applied (which can make sense in some cases) you lost that fine control completely. It's a long process of previewing and then moving repeatedly to check it's accurately placed. 

I've attached an example where the edges need lining up and there are white gaps showing but when I select to move the red object it completely hides the gap making it impossible to know accurately.

**Without Move selected**
[image]

**When move is selected:**
[image]

**Describe the solution you'd like**
A way to turn this outline off, I can see circumstances where this makes sense to see but for smaller objects it's really a hinderance.

**Describe alternatives you've considered**
There's a slight workaround but it's a little jittery - where you select the rotate and then grab the object but more often than not you start rotating it. 

**Additional context**
No further context - happy to provide if more is needed!

rob180sx, 2025-06-05:
Ya this is terrible. Orca has a toggle for this outline and it’s off by default. We need options for these features.

davidtranier-cmyk, 2025-08-12:
This is a real step backwards, how can I make use of the level of detail offered by a 3d printer for tight tolerances if I can't place objects accurately? :|

barneyvaughan, 2025-08-12:
@davidtranier-cmyk It's a shame this has been open for so long, but I don't know how to help give it traction. 

I gave it a go the other week to fix it up, I managed to get the setting into the setting menu in the app, but I couldn't find the time to rig it up to actually turn the outline on and off. Maybe I'll get a minute this month if it's not mentioned by Bambu to raise a PR with it in place.

Nysplaidame, 2025-09-15:
Nothing, I've not updated my firmware since Bambu decided to lock things down so Orca still works for me. At this point in time I don't think I'll bother with Bambu Studio much longer.