bambulab/BambuStudio #6920: build plate just keeps rotating when i open bambu studio after the last update

progfool70, 2025-05-14:
### Bambu Studio Version

20354

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

a1

### How to reproduce

open bambu studio it start rotatating until i click the plate

### Actual results

i open bambu studio and the build plate with the project on it just starts spinning, it happens on every project i open not just certain ones

### Expected results

started after latest update

### Project file & Debug log uploads

all projects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-14:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

progfool70, 2025-05-18:
you have to use the left mouse button and hold it to rotate the screen when i open up bambu studio my screen is all ready rotating i have to hit the left mouse button to stop the rotation then it works normal

Haidiye00, 2025-05-19:
Hello, can this bug be stably reproduced? Is there any relevant operation

progfool70, 2025-05-19:
Thank you I fihured it out …when I use tinkercad then bamboo studio, bamboo studio acts crazy until I restart my computer and it happen every time I go from tinkercad to bamboo weird huh lol

Haidiye00, 2025-05-20:
Okay, if you can provide relevant GIF, we will analyze them. As you said, it has been resolved and we have temporarily closed the issue~

closed 2025-05-20 by Haidiye00

Haidiye00, 2025-07-18:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~