bambulab/BambuStudio #6921: Bulk Editor for Parts

brianfreychet, 2025-05-14:
**Is your feature request related to a problem? Please describe.**
I have models where the part is changed into a negative to be removed from the main part.  I waste a lot of time changing these into negatives, as right now I have to change each individual part with the prompt.

**Describe the solution you'd like**
The ability to bulk edit parts.  Right now you have to select each individual part and click 'Change Type', however it would be extremely beneficial to be able to set the type on multiple parts at once.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Single Piece Context Menu (With Bulk edit option):
[image]

Multiple Part Context menu (No Bulk edit option):
[image]

vanowm, 2025-06-08:
In addition to the request, if anything is removed from dropdown menu, the remaining items should have persistent order.

Currently menu items are all over the place, like `delete` item in one menu is first, in another it's 3rd item, `center` is first in one and 5th in another..

vanowm, 2025-06-08:
Duplicate of [#1400](https://github.com/bambulab/BambuStudio/issues/1400)