bambulab/BambuStudio #6924: Bambu Lab 0.4mm HF nozzle

cmontis, 2025-05-14:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

OS Name	Microsoft Windows 11 Pro Version	10.0.26100 

### Additional system information

OS Name	Microsoft Windows 11 Pro Version	10.0.26100 Build 26100
System Model	Precision 3460
System Type	x64-based PC
Processor	Intel(R) Core(TM) i9-14900, 2000 Mhz, 24 Core(s), 32 Logical Processor(s)
BIOS Version/Date	Dell Inc. 3.15.0, 3/24/2025
BaseBoard Manufacturer	Dell Inc.
Platform Role	Desktop
Installed Physical Memory (RAM)	64.0 GB
Name	NVIDIA RTX 4000 SFF Ada Generation
Driver Version	32.0.15.7159

### Printer

Bambu Lab H2D

### How to reproduce

I connected my new Bambu Lab 0.4mm HF nozzles.
I synchronized the nozzles at the printer....then nothing changes....
I went to Bambu Studio and synched and again nothing....still showing standard nozzles.
I went to the device page and it shows standard nozzles.

Are these actaully HF nozzles?  It says so on the box and the nozzles.

When I manually set it to 0.4mm High Flow on the Prepare tab in Bambu Studio....nothing updates for the PLA Matte material for the new volumetric values.  I would expect similar values to what I was using for E3D (Bambu official partner) on my X1C but there is no change.

Was this a waste of my money or is there an update I am missing?

I found a way on the printer to update the value manually.

### Actual results

Nothing would work so I rebooted the printer and then when I manually set the nozzles to HF it held.  I went to Bambu Studio and synched and it would not match.  I manually set it and told it to ignore synch.

Then I went to each filament and selected HF in the edit and  then saved it as a custom profile.

### Expected results

I expect that Bambu Lab H2D to successfully read the nozzle size, hardened steel, and High Flow.

### Project file & Debug log uploads

[Sparta Nameplate Team.zip](https://github.com/user-attachments/files/20214099/Sparta.Nameplate.Team.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

hotellonely, 2025-05-15:
you have to change it in the printer's settings, not in Bambu Studio. BS should only be used to read (sync) from the printer.

[image]

You also would have to choose the nozzle in profile page to see the profile used for high flow nozzle. PLA matte is using 48mm^3/s limit which is pretty high.

[image]

cmontis, 2025-05-15:
I had to do it both places....I tryed to read the nozzles at the printer....failed to recognized High speed so I clicked edit and set it.The. i tried to get BStudio to synch with printer and it could only synch the nozzle and not the High Speed.  This means I have to set it manually.  And so when I go to slice then I have to tell BStudio to not synch because I am using manual settings.What good is the synch/read function on the printer to read the nozzle if it cannot detect HS?What good is there to have a synch with printer in BStudio is it keeps resetting to normal nozzles.I get the filament settings but I have to manually edit and select High Speed as well.This sounds like a bug if you sell a product and all these settings that are supoae to be automated are forced manually.Dies Bambu Lab plan to fix it.  I cant keep setting everything manually on every job in BStudio.  At least the printer holds it setting but everything else is manial override because I have to synch with printer to update filament reads because I use different colirs on different jobs.Very unhappy but ok...this is how it works!Alex MontisSent from my iPhone

image.png (view on web)
image.png (view on web)

cmontis, 2025-05-15:
Magically, this fixed itself this morning but they could do better with the software.

Last night, I set it in the used the printer to auto-dectect the nozzle.  It failed identify they were HS nozzles so I had to manually edit the nozzles on the printer.  After rebooting, I also rebooted the computer and started Bambu Studio again.

In Bambu Studio, I select synch with printer....that should be all that is required.  It should automatically synch to the information from the AMS and the Printer.  The Device tab was reading standard nozzles but I created my print anyways and printed and with manually adjusted nozzles for HF I was able to print in Ludicrous mode without issue.

This morning, I booted up and you can see below on the device tab it shows that it is reading the printer information correctly.  I still have to manually go into PETG HF and set it but PLA Matte auto sets the Direct Drive High Flow functionality.  **i read that Bambu Lab is only doing it automatically for PLA Matte and PLA basic.  Is that true?**  The PETG HF setting for Direct Drive High Flow is a drop down in the filament setting but you currently have to do it every time but it is very similar to the settings for E3D.  I created filament profiles for PETG HF and PLA Matte so I can visually see that I have my settings loaded without having to remember if I applied it or not since there is a question in my mind on if Bambu Studio did or did not make the adjustment.

**In my mind,  Bambu Studio should load automatically under the filament that uniquely identifies if it is using Direct Drive High Flow** which would help users know if there is a profile setting.  It should not be a hard thing to do on thier end.  After all, we are buying Bambu Lab filament for the unique material identification it offers (in addition to the high quality of hte material and tuned settings for it).

[image]

changbaishen, 2025-05-16:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]