bambulab/BambuStudio #6925: Slicer error in 2.0

tniles, 2025-05-14:
### Bambu Studio Version

2.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.7

### Additional system information

CPU: Apple M2
Memory: 8GB

### Printer

Bambu X1C

### How to reproduce

This is a regression (seems like in the slicer) which was introduced in 2.00.01.50 and does *not* occur in 1.10.01.50 of Bambu Studio (each as downloaded from the official Bambu website).  It seems to be a difference of how the slicer generates the raft or tree (auto) supports.  I was using PETG-HF filament and did not try any other filaments.  

With either version of Studio, I use the exact same STEP file for the part I'm printing, with the exact same printer parameters, and set the printer to print it.  

### Actual results

By the time the printer is done printing the part, the part printed with 1.0 is as expected (no defects, raft/supports as expected, everything is good), whereas the part printed with 2.0 has a different raft/support interface which seems to be the cause for some layer defects in the printed part, occurring in the next several layers above where the support leaves off, making for a bad print every time.  Please see photos. 

Specific to point out: the 1.0 Supports have a thatched or "criss-cross" two layers of raft/interfacing, whereas the 2.0 Supports have only a single layer of a single continuous line. 

[image]
[image]
[image]
[image]
[image]

### Expected results

When printing with 2.0, I would expect the part to print equally as well or better than when printed with 1.0 Studio.  Specifically, with no layer defects which appear to be downstream of differences in the support/support interface.  See photos.  I further expect there to be no differences in the supports and their rafts/interfaces. 

### Project file & Debug log uploads

Project files attached. 

[Proj-Files.zip](https://github.com/user-attachments/files/20214763/Proj-Files.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

tniles, 2025-05-14:
Hmmm... not sure the photo names went through (after upload).  Clearly, there's a couple of bad prints (showing two related but slightly different failure modes), and one good print.  Same for the supports.  Note that I have plenty more of both bad and good prints, all representative of what is pictured in this report.  

Here's what may be the key, which I included in my original report:

_Specific to point out: the 1.0 Supports have a thatched or "criss-cross" two layers of raft/interfacing, whereas the 2.0 Supports have only a single layer of a single continuous line._

tniles, 2025-05-14:
Also to note: ultimately, until this is fixed, I plan to continue using 1.0 of Bambu Studio for as long as I can because print defects like this make the prints unusable.

NothFie, 2025-05-15:
@tniles Thanks for your feedback. Sorry this is indeed a BUG and we are looking into it. We will inform you as soon as it's fixed

tniles, 2025-05-28:
Thanks, @NothFie.  Just following up, is this indeed a slicer bug or is it somewhere else?  Asking to clarify whether adopting a different slicer would resolve the issue.  Not my first choice, as it is easier to keep using Studio 1.0 until this is fixed.

NothFie, 2025-05-29:
@tniles As you said, the interface should have two "criss-cross" layers, this issue is fixed and will be released very soon. Besides, the only difference between 1.0 and 2.0 is that the overhang speed slows down a little (I don't think this would have any negative impact).

NothFie, 2025-06-03:
@tniles This issue should have been fixed in the latest version Studio V2.1.0, please have a check.

NothFie, 2025-06-06:
@tniles It seems that the decrease in speed of overhangs actually led to failure. This is a bit counter-intuitive, we will conduct further testing and get back to you.

tniles, 2025-08-27:
[image]
[image]
[image]
[image]
[image]

tniles, 2025-08-27:
As you can see, v2.2.1 cuts corners on the third wall loop in multiple locations, as though it doesn't know how to follow a curve properly.  This is most noticeable on the inner circle, where instead of a smooth circle (like in the 1.10 print) there is more of a pentagon/hexagon shape.

fahaaaxiki, 2025-08-27:
Got it, We have found that the problem is caused by excessive adjustment of the following overhang speed parameter. It was 50 in version 1.10, but was changed to 10 in version 2.0, causing the wire to be carried away by the nozzle.

[image]

At present, we are verifying the fix, but it will take some time for testing

tniles, 2025-08-27:
@fahaaaxiki thanks for sharing that screenshot!  I did a couple more test prints after changing that setting.  It resolved the wall loops issue (they look clean now, like in the 1.10 print).  That is a very odd setting to change without a lot of testing first--why did this parameter get changed in the first place? 

Comparing between 1.10 and 2.2.1 for PETG-HF settings in the 0.30mm Standard @BBL X1C 0.6 Nozzle profile, I only found two other settings that were also changed.  However, these did not seem to generate any appreciable effects.  These were two Support settings, under Advanced, "Initial layer expansion" and "Support wall loops".  In 2.2.1 they default to Auto, however in 1.10 they are '2' and '0', respectively.  Again, after a couple of test prints, no appreciable differences from what I could tell. 

There is one slight issue still remaining, and I'm hoping someone can point me in the right direction for tweaking settings or if it is a bug then getting a fix out.  I'll share a couple more photos, and what you'll notice is there is a definite "band" or artificial gap in the layers just above where the support interface ends.  If you look really really hard, you can kindof see this in the 1.10 print.  In the 2.2.1 print, it is rather pronounced and easy to see.  Any ideas what would be causing this?  Thanks!

tniles, 2025-08-27:
Notice the "band" that goes ALL the way around the print.  I've had some test prints that end up causing some layer separation in this area (mid-print).  However, the overhang speed fix seems to have helped that greatly.  But this is still a definite quality/print issue introduced somewhere after 1.10.  Thoughts?  Clues?

[image]
[image]

fahaaaxiki, 2025-08-28:
1. Modifying the Overhang speed parameter is intended to adjust the printing quality of models with some overhangs, but it seems that it has also brought about some negative effects, just like the problem you discovered
2. 2. We will check again whether the "band" issues are bugs or not. Thanks