bambulab/BambuStudio #6937: Selecting global process crashes app

DurrtySteve, 2025-05-16:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Mac 15.3.2

### Additional system information

MacBook Pro M1

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Open app
2. Open a Recently Opened project
3. Under Process, select Global
4. Select any Process from the dropdown

### Actual results

Application crashes.

### Expected results

The selected Global Process should be applied.

### Project file & Debug log uploads

Files are failing to upload!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-16:
Thank you for your feedback. Could you please provide the log file? The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

Haidiye00, 2025-07-18:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~