bambulab/BambuStudio #6938: Flow rate calibration does not heat up the bed

aphex008, 2025-05-16:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Macos

### Printer

H2D

### How to reproduce

1. Go to "Calibration"
2. Start "Flow rate calibration" using ABS with 65 degrees chamber temperature and filament preset of heat bed 100 degrees 

### Actual results

Printer is waiting for chamber to heat up to 65 degrees but the bed heating is off.
Heat up takes very long time because printer is using only chamber heater but the bed is cold.

### Expected results

Calibration should start heating bed and chamber simultaneously so it takes shorter time.

### Project file & Debug log uploads

[flow_rate_fine_calib_mode.3mf.zip](https://github.com/user-attachments/files/20239611/flow_rate_fine_calib_mode.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

gaolegao-lx, 2025-05-19:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

rabacen, 2025-09-01:
This happens on the A1 when running Flow rate calibration. Bed does not heat up. Only way to run the test is to manually set the temp after starting calibration.