bambulab/BambuStudio #6940: No "Use AMS" Option When Sending To Print Bed/Printing From SD Card

whatwouldlewdo, 2025-05-16:
### Bambu Studio Version

v2.0.3

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1 w/AMS Lite

### How to reproduce

1) Slice Multi-colored model.
2) Choose "Print Plate."
3) "Use AMS" is no longer an option.
4) Export and attempt to print it from the SD card.
5) No option for AMS there either (only newly exported models effected).

### Actual results

The printer will use only one color for everything.

### Expected results

The printer should print all the colors...

### Project file & Debug log uploads

[New Compressed (zipped) Folder.zip](https://github.com/user-attachments/files/20242327/New.Compressed.zipped.Folder.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-05-17:
- Send Print the Plate:
https://github.com/user-attachments/assets/4e9fd15c-32f2-44f7-8c22-ddda865a81f8

- Print from SD card:

[image]

The above two gifs shows the progress of my A1 with your 3mf and it worked well with Studio V2.0.3.54. I don't know if my operation   was different from yours. If this problem exists, can you upload your opeartion screenshot or gif to show the progress. Thanks for your feedback again.

whatwouldlewdo, 2025-05-17:
Do you not see that the "Use AMS" option is gone ? There was a box you check before you sent it through, and it's not on yours either.

If you save this to an SD card, then go to print it from there, you should also see it missing from the screen within the printer as well. If you DO see it, then there is clearly an issue somewhere on my end. I didn't have any issues until the latest update. 😏 

What's weird though is that the option IS present on the A1 screen with models that were exported prior to this issue AND models that only have one color. "Use AMS" is right there at the top, but for some reason, not with multicolored prints. 🤷🏻‍♂️

JohnSmithCYM, 2025-05-17:
We have removed the "Use AMS" option from V2.0.2, if you want to print with AMS or exteranl spool, you can select the filament like I show in the gif. The same logic is on firmware of H2D and new X1C now. Thanks for your feedback and you can tell me if there is anything unconvient after we change this logic.

whatwouldlewdo, 2025-05-17:
I noticed that you send it to plate from the device page, not the "prepare" section. That's where I've been sending it to the plate, but again, even when I try to print it from the SD card, I still have no "use AMS" option. How would you explain that ?

JohnSmithCYM, 2025-05-19:
We have removed "Use AMS" option from V2.0.2 in Send print dialog, which is reused both in "Print Plate" in prepare/preview tab or in SD card print dialog.

whatwouldlewdo, 2025-05-19:
It's on the screen of the printer, not the slicer. It STILL very much has "Use AMS". I will show you...

whatwouldlewdo, 2025-05-19:
[image]
[image]
[image]

The photos you see that have the "Use AMS" (like I told you they do) are single color except for the last photo. That is the last version of the multicolored logo that I exported before the update. NOW no matter how many times or how many ways I export it (even starting from scratch), I cannot get it to print in anything but black. Yet why does a single color print give me that option ? I don't know what the problem is, but it's really frustrating and I need a solution ASAP. If you actually printed the file, you would see the same results...