bambulab/BambuStudio #6943: No Fan control after Studio/Firmware upgrade

stevek88, 2025-05-16:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel(R) Core(TM) i7-10700K CPU @ 3.80GHz   3.79 GHz
64.0 GB (63.8 GB usable)
Windows 11 Home
23H2

### Printer

X1C

### How to reproduce

During print go to Device tab
In control window click on Fan control icon.

### Actual results

Instead of menu and options to control chamber fans message appears without any options for fan control

### Expected results

Options to control fans

### Project file & Debug log uploads

Not project related.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

changbaishen, 2025-05-20:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

stevek88, 2025-05-20:
I don't understand what log you want.  I click on fan icon in bambu studio.  Used to get a menu to control printers.  Now just a message without control options.  Seems like it's an issue with new Bambu studio/printer firmware version.  This feature needs to be brought back

Kams84, 2025-05-20:
https://github.com/user-attachments/assets/2a41b715-9323-4084-b7ff-4861349c0a79
I have same issue.
Also, speeds displayed in Handy are all wrong.