bambulab/BambuStudio #6944: Fan control missing after SW update

stevek88, 2025-05-16:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 PC

### Additional system information

Intel(R) Core(TM) i7-10700K CPU @ 3.80GHz   3.79 GHz
64.0 GB (63.8 GB usable)
64-bit operating system, x64-based processor

### Printer

Bambu Lab X1C

### How to reproduce

start print
go to Device tab in bambu studio.
In control sections click on Fan icon

### Actual results

Menu appears with message:
Cooling mode is suitable for printing PLA/PETG/TPU materials and filters the chamber air.

### Expected results

should show controls to change fan settings.  THIS IS A MUST!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

### Project file & Debug log uploads

project independent

### Checklist of files to include

- [ ] Log file
- [ ] Project file

changbaishen, 2025-05-20:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

vreichl, 2025-05-25:
"should show controls to change fan settings. THIS IS A MUST!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!"

fully agree to that statement. Why the fan settings have been removed? I am sure, that this was not a proposal from the engineers but rather something coming out of the disturbed minds of some decision makers within the company (knowing big companies myself as well...).

Please do not hesitate to reintroduce FAN-control in Bambu studio!!