bambulab/BambuStudio #6951: Mesa 21.3+ change effects Bambu Studio

YoSTEALTH, 2025-05-16:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Built from source by yourself or others

### OS version

Linux 6.14.6-2-cachyos (64-bit)

### Additional system information

X11

### Printer

A1

### How to reproduce

With `Mesa 25.1.0` installed, launching `bambu-studio` will say `/usr/lib/libOSMesa.so.8` not found.

According to https://docs.mesa3d.org/amber.html "After Mesa 21.3, all non-Gallium DRI drivers were removed from the Mesa source-tree."

My current solution was to uninstall `mesa` and install `mesa-amber` for legacy support.

Bambu Studio will need to add mesa 21.3 version check and account for new changes.

### Actual results

none

### Expected results

none

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jipiboily, 2025-05-18:
Having the same issue!

Bambu printing on Linux has been a pretty bad experience.

Hope this can be fixed quickly!

jipiboily, 2025-05-18:
The flatpack version seems to work, btw!

MackBambu, 2025-05-20:
Thanks for the reminder ， I’ll try adding some additional prompts.

c2h5oh, 2025-05-23:
Mesa 25.1 removes osmesa completely - this will have to be fixed sooner than later.

fuzunspm, 2025-05-24:
same issue on arch linux after latest mesa update, flatpak version works but this should be fixed

princess-wawa, 2025-05-24:
// copied because i posted it on a duplicated issue

basically, libOSMesa.so.8 is no longer included in the Mesa package since version 1:25.1.0-1
see the merge request: https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/33836

this means that libOSMesa, which bambustudio relies on, has just been removed from the Mesa package for no apparent reason other than "redundency"

note that my graphical interface is gnome, and since it has been updated to use the new Mesa version, downgrading won't work, and the mesa-amber doesn't work with it either

the fix is fortunately very simple, either switch to the flatpak version since it still uses the old mesa version that includes osmesa
or replace your mesa package with the amber branch since it still has the osmesa lib

fuzunspm, 2025-05-25:
This is not a fix but a temporary workaround.

Knifa, 2025-05-25:
`mesa-amber` is missing `zink` for the NVIDIA workaround #4669 so the only option is to downgrade in that case.

rjuarezp, 2025-05-25:
Please update the libOSMesa library! Solus linux has updated to Mesa 25.1 and now I cannot use Bambu Studio anymore under my Linux!

c2h5oh, 2025-05-25:
There is another option: remove it completely from dependencies and BambuStudio should still work. 

I package OrcaSlicer for Arch and I patch it out before building and there don't seem to be any ill effects of that. Here's a PR for Orca https://github.com/SoftFever/OrcaSlicer/pull/9708 that most likely can be copied verbatim into BambuStudio.

Meissullo, 2025-05-26:
can confirm orca slicer still works on arch with hyperland,  bambustudio does not.
Used the nvidia workaround [#4669](https://github.com/bambulab/BambuStudio/issues/4669) on both

Nordaar, 2025-06-02:
For arch+nvidia users: 
the aur `bambustudio-nvidia-bin 02.00.03.54-4` package now contains all the missing libs and works with mesa upgraded past `25.0.5`.
Tested with:
- `linux 6.14.9.arch1-1`
- `bambustudio-nvidia-bin 02.00.03.54-4`
- `mesa 1:25.1.1-2`
- `wayland 1.23.1-2`
- `hyprland 0.49.0-1`
- `nvidia-dkms 570.153.02-1`

david-barbion, 2025-06-04:
BambuStudio seems to have no use of OSMesa but with linked with it, so it requires it to start. A dirty hack is to fake libOSMesa by symlinking with, for example, libGL which gives, on Arch:
```
sudo ln -sf /usr/lib/libGL.so /usr/lib/libOSMesa.so.8
```
then start bambustudio as usual

bedfelllow, 2025-07-29:
Flatpack is not a good option for me on Fedora 42 KDE, I as it regularly crashes.

hadess, 2025-08-16:
> Offscreen rendering methods require a dependency on OSMesa

The correct fix for this would be to use EGL and render to the "software" device:
https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/33836#note_2806856

I should note that in the PR above, I assessed that the offscreen rendering probably always used X11 anyway, as it was initialising X11 thread support earlier.

hadess, 2025-08-16:
> Flatpack is not a good option for me on Fedora 42 KDE, as it regularly crashes.

The code in the Flatpak is the same that runs in AppImage, so this is unlikely to be related to the Flatpak itself. You should definitely file bugs against Bambu Studio here when they happen.

hadess, 2025-08-16:
> The correct fix for this would be to use EGL and render to the "software" device:
> https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/33836#note_2806856

Example offscreen render in glfw:
https://github.com/glfw/glfw/blob/master/examples/offscreen.c

hadess, 2025-08-22:
> I did some checks: when building GLFW on Linux, the build selected GLFW_USE_WAYLAND=ON, so GLFW_USE_OSMESA wasn’t set. I also inspected the AppImage executable with ldd, which confirms that linking against OSMesa didn’t actually take effect. If GLFW_BUILD_X11 is enabled, will it cause OSMesa to be linked?

OSMesa is added to the list of libraries to link the main binary to in `src/slic3r/CMakeLists.txt`:
```
elseif (CMAKE_SYSTEM_NAME STREQUAL "Linux")
[...]
    target_link_libraries(libslic3r_gui ${DBUS_LIBRARIES} OSMesa)
```
It's not a dependency that's dragged in by GLFW.

About GLFW itself, you shouldn't be using the `GLFW_USE_OSMESA` build flag as it's been removed. From the `CMakeLists.txt` in GLFW:
```
if (GLFW_USE_OSMESA)
    message(FATAL_ERROR "GLFW_USE_OSMESA has been removed; set the GLFW_PLATFORM init hint")
endif()
```

The OSMesa support will still be there in the default build, but only if you link against OSMesa... So I'm pretty sure that OSMesa is still available in GLFW, and linked to the main binary.

About the commits that are referenced above, adding OSMesa as a private dependency is definitely not the right fix for this, the right fix would be to remove the obsolete OSMesa dependency and use EGL and a "software" device to render to when running without a windowing system.