bambulab/BambuStudio #6954: Ability to hide objects from view without removing them (eg. modifiers)

IgnusG, 2025-05-17:
**Is your feature request related to a problem? Please describe.**
I'm using large wood log modifiers to add texture and wood patterns to my wood filament prints. The idea is that this modifier object covers the entire item to produce the pattern everywhere. As a consequence though it completely overshadows the item and makes it impossible to click on it (eg. to move, rotate, paint supports etc.).

If printing multiple items the situation gets even worse.

Here's an image of the Prepare view with the modifiers visible. You can barely make out the item itself and without moving the modifier it's not possible to edit it:

[image]

Here's how it looks sliced (since the Preview panel does not show modifiers anymore):

[image]

**Describe the solution you'd like**
A simple button in the object menu to hide specific items (like modifiers) or groups of items. This is very similar to other software products like Fusion or Photoshop where hiding different solids/layers is possible to focus only on one specific one.

[image]

Slicing should always include all items regardless whether they are hidden from view or not (not the same as the checkmark which excludes them entirely). 

**Describe alternatives you've considered**
What I have to do now is each time I need to edit the item itself I must move the modifier out of the way and then move it back.

JonLaliberte, 2025-05-23:
Yeah, this and locking.
As implemented this feature is pretty useless - it doesn't actually hide the object, nor does it stop you from moving it.

Beeblebrox666, 2026-09-28:
I agree this would be a life saver, even Tinkercad lets you hide parts, then bring them back later without moving their coordinate location.

This should be a critical fix it yesterday idea...