bambulab/BambuStudio #6956: Unknown error

kuncl182, 2025-05-17:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

I dont know.. It just dostn't want to start printing .. I hate your printer more and more - just because of your software !

[image]

### Actual results

WONT PRINT

### Expected results

FIX BUGS . 

### Project file & Debug log uploads

>...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-19:
Sorry, could you please update to the latest version V2.0.3.54 on the official website and then try again? Download link: https://bambulab.cn/zh-cn/download/studio

kuncl182, 2025-05-19:
> @kuncl182 <https://github.com/kuncl182> Sorry, this issue has been fixed
> in our latest version. Could you please update to the latest version
> V2.0.3.54 available on the official website and try again? Download link:
> https://bambulab.cn/zh-cn/download/studio

No i don't want to update my bambu studio. Because i allready have one
computer with the latest version of bambu lab studio and im not sattisfies
with the printing proces ( poor addishion on the supprt layers, printing
speed, bad adhision ...) so that's why i want to use this older version of
slicer because it works like i want it to...

BoBoYu-BamBu, 2025-05-20:
@kuncl182  Oh, I see. My suggestion is that you can try this issue on a new version of another computer. After all, we have fixed it in the new version update. Thank you.

closed 2025-05-22 by BoBoYu-BamBu