bambulab/BambuStudio #6957: H2D Ludicrous Broken

cmontis, 2025-05-17:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

OS Name	Microsoft Windows 11 Pro Version	10.0.26100 

### Additional system information

OS Name	Microsoft Windows 11 Pro Version	10.0.26100 Build 26100
System Model	Precision 3460
System Type	x64-based PC
Processor	Intel(R) Core(TM) i9-14900, 2000 Mhz, 24 Core(s), 32 Logical Processor(s)
BIOS Version/Date	Dell Inc. 3.15.0, 3/24/2025
BaseBoard Manufacturer	Dell Inc.
Platform Role	Desktop
Installed Physical Memory (RAM)	64.0 GB
Name	NVIDIA RTX 4000 SFF Ada Generation
Driver Version	32.0.15.7159

### Printer

Bambu Lab H2D

### How to reproduce

Load the file
Preview the file  (observe it takes 1.5 days to print)
Notice that I have 0.4mm HS profile loaded for the nozzle and for the filament).
Print the file
Select at any time but I do it have startup calibration to move up to ludicrous speed.
Observe that the time drops to one day (26/1.66=~24hours).

### Actual results

Per the screenshot I am in ludicrous mode and it took 36 hours.  It continually lied until it could not meet the time and then updated each time.  I essentially printed in standard printing.

### Expected results

I expect that when the system confirms the time is reduced to 24 hours that this is reality.

### Project file & Debug log uploads

[RubicsWhiteLampALT_1.zip](https://github.com/user-attachments/files/20266377/RubicsWhiteLampALT_1.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

gaolegao-lx, 2025-05-20:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

cmontis, 2025-05-20:
I went to Handy, then Me, then ticket service, ticket history...and I find
only two open tickets and this one is not found...
6957 is not found.

Create a new one?

cmontis, 2025-05-20:
I created a new incident because I could not find this one in my tickets.

New ticket No:
US250520519001

The printer is in the middle of a job but I will append the log as soon as
I get home from work.  I thought I could pause to upload but it has to be
idle.