bambulab/BambuStudio #6962: Bambu Studio Threads Bug - Adding Extra Threads Where They Do Not Exist

Nikibahama, 2025-05-18:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

CPU:  Ryzen Threadripper 3970X 32-Core @ 3693 Mhz
RAM:  128GB DDR5
GPU:  Nvidia RTX 3090 with 24GB VRAM

### Printer

Bambu Lab P1S

### How to reproduce

Open the file, position it where the threads are horizontal to the build plate, click "slice plate".  

Next: go back to Preview, turn the object 90 degrees so the threads are perpendicular to the build plate and click "slice plate".  

[image]
[image]

### Actual results

Whenever the threads are horizontal (parallel to the build plate), the software is glitching and adding in extra material that does not exist in Solidworks, or the Bambu Studio Preview Window (or in any other slicer for that matter: Simplify 3D, Cura, Anycubic, Chitubox).  This then leads to print failures and problems with horribly-recreated threads thanks to Bambu Studio.  

Whenever the threads are vertical (perpendicular to the build plate) the software displays the part correctly and prints correctly.  

It doesn't matter if I create the threads manually in Solidworks or if I use the Solidworks thread wizard tool.  It doesn't matter if I cut the threads or extrude the threads.  The problem exists only in Bambu Studio with the resulting STL files (not in any other slicer).    

### Expected results

The software should be displaying the threads correctly no matter how the part is positioned.  It should not be adding in extra threads outside of the physical object that do not exist.  

### Project file & Debug log uploads

[Bug Files.zip](https://github.com/user-attachments/files/20273606/Bug.Files.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Nikibahama, 2025-05-18:
This problem exists in both V2.02 and V2.03 due to inner and outer wall line width settings.  I narrowed down the cause today.  When imported under my profile with a 0.42 outer wall and 0.48 inner wall, the models always slice with extra material on the threads and alternate material missing from the model (if the threads are parallel to the build plate).  If I then change my inner/outer wall to 0.45mm each and re-slice, the glitch disappears.  If I then change my settings back to 0.42/0.48 after this step and re-slice, everything slices as it should and prints perfectly; however if I add any new parts onto the plate or import another set of parts under the 0.42/0.48 settings, the glitch returns.  I have to again repeat the 0.45/0.45, re-slice, then 0.42/0.48 and can print just fine...  So this is most certainly a glitch in the software and I've managed to determine exactly what settings are causing it.  Time for Bambu Lab to fix their Studio software.

QingZhangBambu, 2025-05-22:
[image]
Hi! The layer height is set to 0.2mm, however, the scarf seam start height is set to 2mm. This will result in the loop start position being much higher than it should be.