bambulab/BambuStudio #6963: Bambu studio no me deja ver la rueda de ajustes de filamentos, en filaqmentos del proyecto, en la sección preparar. No puedo añadir filamentos, no puedo ver la rueda dentada mas a la derecha para acceder a ajustes.

juan6627, 2025-05-18:
### Bambu Studio Version

Bambu Studio 2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Todas las series

### How to reproduce

[image]

### Actual results

nada

### Expected results

nada

### Project file & Debug log uploads

nada

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-19:
[image]
Hello, I suggest you adjust the resolution or zoom ratio of Windows~

juan6627, 2025-05-19:
[image] Hola y gracias por la ayuda, pero no se soluciona y he probado diferentes configuraciones. Como puede apreciar en la imagen tengo la resolución recomendada y escala normal, pero el problema persiste. un saludo.

Haidiye00, 2025-05-20:
Hello, it seems that the text on the left is too long. Let's consider how to improve it. I suggest you use the English version first~

juan6627, 2025-05-20:
[image] Efectivamente la versión en ingles me abre el programa correctamente, gracias. Espero que lo implementen ya que no hablo ingles y comprender el laminador en otro idioma es mas complicado para su comprensión. Reciba un cordial saludo.

juan6627, 2025-07-26:
> https://github.com/bambulab/BambuStudio/releases
> Hello, please use the latest beta version. Thank you~

Con la versión beta me va perfecto en español. Además me encanta que se pueda ajustar el sidebar o ventana de ajustes como yo prefiera. Gracias por todo .

Haidiye00, 2025-07-28:
ok，thanks

closed 2025-07-28 by Haidiye00