bambulab/BambuStudio #6966: Infill knocks print off bed

KarisMajik, 2025-05-19:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Irrelevant, this is printer-side

### Printer

A1

### How to reproduce

Use grid, adaptive cubic, or any other infill where the nozzle will cross over previously printed parts of the infill from that same layer, repeatedly, on taller prints. The first time It happened was near the end of a print that was about 10cm tall. Twice, same file. It just happened again with something the size of my thumb

### Actual results

Eventually, the nozzle crossing over infill it just printed earlier that same layer

### Expected results

The print to stay on the bed and finish in one piece. Maybe *don't* have grid for any presets

### Project file & Debug log uploads

There's literally nothing file-wise to upload, this is not a programming issue, it's a printing issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-19:
@KarisMajik  Thank you for your feedback. Could you please provide the printed screen recording and the 3mf file?

KarisMajik, 2025-05-19:
With all due respect, this is an issue that has been known for over a year, I shouldn't have to do more work for you to fix your product, when you can look it up yourself, like here for example https://www.reddit.com/r/BambuLab/s/qaxwRFfG3u

JohnSmithCYM, 2025-05-20:
Thanks for your feedback and I have tried it with my A1  mini & A1 +PLA Basic, I printed a tall cube (more thant 150mm and preset is in 0.2mm Standard) and it works well. I read the reddit you offer and the default Sparse Infill is the same with my condition, and as you mentioned, I didn't open Timelaspsed in my work.
Can you offer just one example 3mf for me to give it a further test? Due to I can't reproduce it.

KarisMajik, 2025-05-21:
[image]

BoBoYu-BamBu, 2025-05-21:
@KarisMajik Oh, would it be convenient for you to directly send the 3mf file to my email: yuting.huang@bambulab.com? Thanks