bambulab/BambuStudio #6970: Can not start print with a custom filament from bambu studio

TheCraftDude, 2025-05-19:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Business 20H2

### Additional system information

Lenovo Thickpad L480
CPU: intel i5-8250U
Ram: 8GB

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1: create new project
2: import STL file
3: under "Project Filaments" click "Set filaments to use"
4: click "Custom Fillaments"
5: click "Create New"
6: click "Can´t find vendor i want"
7: Enter name of vendor under "Input Custom vendor" ( 3DE )
8: under Type select "PETG-CF"
9: Enter color under Serial ( PBT+ Snow White )
10: under Filament Presets select "Generic PETG-CF"
11: select "Bambu Lab X1 Carbon 0.6 nozzle"
12: click "Creat"
13: click "SLice Plate"
14: click "Print Plate"
15: in the new window, under "Filament" an error is orrucing "Unable to automatically match to suitable filament. Please click to manually match"

### Actual results

it is not plausible to upload a custom filament to the printer, neither is it almost implausible to print a project, when creating or selecting a custom filament, if this specific filament is not on the current vendor list, on the machine or bambu Studio.
After having tried to reconct to wifi, turning it on and off again, deleting the filament, and then factory reset, the only solution i have found to print with a custrom filament, is to delete any presets on the X1 Carbon, do a factory reset, reconnect again, and never setup any filaments on the machine, or else, it will not alow you to print.

### Expected results

Bambu Studio should be able to upload the custom filament directly to the printer, to ensure the settings on the printer and bambu studio is the same, since it is appearing, that the synchronization feature, does simple not work, neither from the mobile app, or from a desktop pc, there is no option to continue anyway, if you are an advanced user or developer

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20282244/log.zip)

[Test Debug custom filament.zip](https://github.com/user-attachments/files/20282247/Test.Debug.custom.filament.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

XinZhangBBL, 2025-05-20:
@TheCraftDude Thank you for your feedback.  We are looking in this issue .  (1) Can not set custom filament.  (2) Can not print from different type material.

thuijzer, 2025-05-20:
@TheCraftDude I had the same problem, but in step 15 you can click on the filament which opens the 'nozzle' popup. And then select the 'external' filament.

I think the idea is that the printer cannot detect the chosen filament and you should confirm that you want to use an 'external' filament.

XinZhangBBL, 2025-06-03:
@thuijzer @TheCraftDude   thank you for your feedback. Please try the latest version.

[image]

XinZhangBBL, 2025-07-01:
Thank you for your feedback and this issue will be closed.
When you encounter any problems, feel free to provide feedback to us.

closed 2025-07-01 by XinZhangBBL