bambulab/BambuStudio #6975: can't send print job with beta version

LazerC, 2025-05-19:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Slice The Model
2. Send To Printer

### Actual results

It shows sending to printer, but when it ends it jumps to the printer screen and nothing happens

### Expected results

Start Printing

### Project file & Debug log uploads

[DeckRooftop Bambu.zip](https://github.com/user-attachments/files/20309147/DeckRooftop.Bambu.zip)

[log_iotc.txt](https://github.com/user-attachments/files/20309650/log_iotc.txt)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-21:
@LazerC  Thank you for your feedback. The log file mentioned above contains no useful information. Could you please zip up the entire log folder as shown in the picture?
[image]

LazerC, 2025-05-21:
[log.zip](https://github.com/user-attachments/files/20358970/log.zip)

BoBoYu-BamBu, 2025-05-21:
@LazerC  Sorry, after our investigation, it seems that some configuration files have been damaged. I suggest you uninstall the software (clear the configuration files), reinstall BambuStudio completely and then try to initiate the printing again.

LazerC, 2025-06-08:
yes, thanks!