bambulab/BambuStudio #6976: Importing a 3MF project model into another 3MF project doesn't import object or modifier settings

jason0l, 2025-05-20:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Series

### How to reproduce

1. Add any model to a new project (a primitive cube is used as an example)
2. Make some changes to the model object, for example:
- Change model object settings
- Add modifier cube to model and change modifier cube object settings
- Add height range modifier to model and change height range modifier settings
- Add negative cube to model
- Add seam painting
- Add support painting

[image]

3. Save the project (saved as "Test_cube_with_modifiers")
4. Create a second new project with any model
5. Import previous project into the second project as "Import geometry only"

[Test_cube_with_modifers.zip](https://github.com/user-attachments/files/20321120/Test_cube_with_modifers.zip)

### Actual results

The model, modifier cube, negative cube and height range modifier are imported into the second project (with a cylinder as an example). The seam and support painting on the model, and height range modifier settings are retained, but model object settings and modifier object settings are lost.

[image]

[Import_test_actual_result.zip](https://github.com/user-attachments/files/20321127/Import_test_actual_result.zip)

### Expected results

Whilst "Import geometry only" would've implied to me that it would only bring in just the STL model, it seems to do more than that such that I would've thought object settings and modifier settings would also be imported.

[image]

[Import_test_expected_result.zip](https://github.com/user-attachments/files/20323205/Import_test_expected_result.zip)

### Project file & Debug log uploads

[all_test_files.zip](https://github.com/user-attachments/files/20323214/all_test_files.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-21:
[image]
Hello, which information are you referring to being lost? Could you please take a screenshot? Thank you. I have a rough look and most of it is now available

jason0l, 2025-05-21:
> [image] Hello, which information are you referring to being lost? Could you please take a screenshot? Thank you. I have a rough look and most of it is now available

[image]

[image]

Haidiye00, 2025-05-21:
Hello, I understand what you mean. It was indeed our negligence. I would like to ask, which modifier settings can cause slices to be cut like yours?

Haidiye00, 2025-05-21:
[image]
Hello, are these 5 parameters

jason0l, 2025-05-21:
> [image] Hello, are these 5 parameters？

Yes, those are the 5 parameters I had changed in the object settings before importing into a second 3MF project. But the issue isn't specific to those 5 parameters. I had tried to change other parameters, but they are still lost when importing. I had change some random parameters in "Quality", "Strength", "Speed", "Support" and "Other", and the result is the same.

Any parameter changes to these 3 object will be lost when imported to a second 3MF project

[image]

jason0l, 2025-05-21:
Here's a test cube I had tried with over 30 different parameter changes. 

[image]

[Test_cube_with_modifiers_2.zip](https://github.com/user-attachments/files/20359664/Test_cube_with_modifiers_2.zip)

Once imported into a second 3MF project, all parameter changes in the 3 objects are lost. Only parameter changes to the height range modifier is retained.

[image]

Haidiye00, 2025-05-21:
Hello, we will conduct an analysis. Thank you for your feedback!