bambulab/BambuStudio #6982: Finished jobs stay at 0 and don't end in app

trich101, 2025-05-20:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 Pro

### Additional system information

Edition	Windows 10 Pro
Version	2009
Installed on	‎1/‎22/‎2024
OS Build	19045.5854

Device Name	DESKTOP-VJD4VDK
Processor	Intel(R) Core(TM) i7-9850H CPU @ 2.60GHz   2.59 GHz
Installed RAM	16.0 GB (15.7 GB usable)
Storage	954 GB SSD PM981a NVMe Samsung 1024GB, 29 GB Realtek PCIE Card Reader
Graphics Card	NVIDIA Quadro T1000 (4 GB), Intel(R) UHD Graphics 630 (128 MB)
Device ID	828372C6-0991-495A-841D-85DF1ABD8749
Product ID	00330-52405-06573-AAOEM
System Type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

A1 Mini

### How to reproduce

Print job.
Job completes.
View printer LCD or APP devices status. The job status has reduced to -0m and even says finished, but doesn't close out. Forcing manual cancel to free up printer for new job.

[image]

### Actual results

Says finished and doesn't close out print

### Expected results

Close out print

This is the 4th or so time its happened on the last few weeks. Different tasks, nothing related to a specific file. Item itself printed successfully and app says finished, just does not move to next step.

### Project file & Debug log uploads

Its not print job failing. Has happened sporadically on 3 other occasions.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-21:
@trich101  Thank you for your feedback.  Could you please zip up the entire log folder shown in the picture and provide it to us?
[image]

BoBoYu-BamBu, 2025-06-11:
@trich101 Hello, may I ask if your problem has been resolved?