bambulab/BambuStudio #6984: ABS Support Material "unknown type"

Scott-Garrett-PWC, 2025-05-21:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5

### Additional system information

M4 Mac Mini

### Printer

Bambu Lab H2D

### How to reproduce

Put official Bambu ABS Support Filament in AMS, and try to sync printer information.  Returns error:

`There are some unknown or uncompatible filaments mapped to generic preset.
Please update Bambu Studio or restart Bambu Studio to check if there is an update to system presets.
- B4(Sup.ABS) The filament model is unknown. Still using the previous filament preset.`

Other people in the forums are having the same issue.

### Actual results

`There are some unknown or uncompatible filaments mapped to generic preset.
Please update Bambu Studio or restart Bambu Studio to check if there is an update to system presets.
- B4(Sup.ABS) The filament model is unknown. Still using the previous filament preset.`

### Expected results

The ability to print with ABS support material.  As it is, it cannot be set to print properly.

### Project file & Debug log uploads

The project is irrelevant.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Scott-Garrett-PWC, 2025-05-21:
Log file of app has 15,000+ lines similar to these since I put the support filament in the AMS this morning:

`debug_Tue_May_20_23_02_57_50000.log.0:2025-05-20 23:57:13.690605[Thread 0x0000000207a61f00]:The values of tray_info_idx and tray_type do not match tray_info_idx GFS06 tray_type ABS-S system_typeSup.ABS
debug_Tue_May_20_23_02_57_50000.log.0:"tray_type": "ABS-S",
debug_Tue_May_20_23_02_57_50000.log.0:2025-05-20 23:57:14.993360[Thread 0x0000000207a61f00]:The values of tray_info_idx and tray_type do not match tray_info_idx GFS06 tray_type ABS-S system_typeSup.ABS
debug_Tue_May_20_23_02_57_50000.log.0:2025-05-20 23:57:16.137682[Thread 0x0000000207a61f00]:The values of tray_info_idx and tray_type do not match tray_info_idx GFS06 tray_type ABS-S system_typeSup.ABS
debug_Tue_May_20_23_02_57_50000.log.0:2025-05-20 23:57:17.379326[Thread 0x0000000207a61f00]:The values of tray_info_idx and tray_type do not match tray_info_idx GFS06 tray_type ABS-S system_typeSup.ABS
debug_Tue_May_20_23_02_57_50000.log.0:"tray_type": "ABS-S",
debug_Tue_May_20_23_02_57_50000.log.0:2025-05-20 23:57:18.534564[Thread 0x0000000207a61f00]:The values of tray_info_idx and tray_type do not match tray_info_idx GFS06 tray_type ABS-S system_typeSup.ABS
`

Scott-Garrett-PWC, 2025-05-21:
That did not fix the issue.  I am, however, on the Mac version, not Windows.  When I removed the system folder in ~/Library/Application Support/BambuStudio, and then launched the app, it prompted me to install BBL 2.0.0.90.  Then, when I try to synchronize the filament with the H2D, I get the same error about unknown filament.

Scott-Garrett-PWC, 2025-05-21:
And just now, I did a complete uninstall of the app, preferences, and support folder, and then re-installed, and it still gives the same error.

BoBoYu-BamBu, 2025-05-22:
Could you please zip up the entire log folder shown in the picture and provide it to us? Thanks
[image]

Scott-Garrett-PWC, 2025-05-22:
[log-6984.zip](https://github.com/user-attachments/files/20378526/log-6984.zip)

Scott-Garrett-PWC, 2025-05-25:
Any thoughts on the issue?  I took another update, but it didn't fix this issue.