bambulab/BambuStudio #6985: Not showing models on Ubuntu 25.04

dah950, 2025-05-21:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Ubuntu 25.04

### Additional system information

Cpu: AMD Ryzen™ 5 7600X × 12
RAM: 32GB DDR5
GPU: RTX 4060
Windowing System: Wayland

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Download and open any file on the Bambu Studio app from Flathub.
See that there is no render of the 3D model.

### Actual results

I can still slice the models but I cannot see the model before and after slicing.

[image]

### Expected results

Being able to see the model before and after slicing.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20360333/log.zip)
[A Benchy at Sea.zip](https://github.com/user-attachments/files/20360334/A.Benchy.at.Sea.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-05-21:
hello,try this 
`sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json ~/
flatpak override --user io.github.softfever.OrcaSlicer --env=__EGL_VENDOR_LIBRARY_FILENAMES=$HOME/50_mesa.json`

guanyun-gudujian, 2025-05-29:
> Hi! I get the following results:
> 
> ```
> ./glcanvastest: line 2: wx-config: command not found
> /dev/fd/63:2:10: fatal error: wx/wx.h: No such file or directory
> compilation terminated.
> Test wayland backend
> ./glcanvastest: line 5: ./glcanvastest.exe: No such file or directory
> Test x11 backend
> ./glcanvastest: line 5: ./glcanvastest.exe: No such file or directory
> ```

[image]

d01, 2025-05-29:
I get:

```
Error: Unable to locate package libwxgtk3.0-gtk3-dev
Error: Couldn't find any package by glob 'libwxgtk3.0-gtk3-dev'
```

Related, just to check, in Ubuntu 25.04, I get:

```
$ apt search libwxgtk
libwxgtk-gl3.2-1t64/plucky,now 3.2.6+dfsg-2ubuntu1 amd64 [installed]
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 gl library runtime)

libwxgtk-media3.2-1t64/plucky 3.2.6+dfsg-2ubuntu1 amd64
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 media library runtime)

libwxgtk-media3.2-dev/plucky 3.2.6+dfsg-2ubuntu1 amd64
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 media library development)

libwxgtk-webview3.2-1t64/plucky 3.2.6+dfsg-2ubuntu1 amd64
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 webview library runtime)

libwxgtk-webview3.2-dev/plucky 3.2.6+dfsg-2ubuntu1 amd64
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 webview library development)

libwxgtk3.0-gtk3-0v5/now 3.0.5.1+dfsg-4 amd64 [installed,local]
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 runtime)

libwxgtk3.2-1t64/plucky,now 3.2.6+dfsg-2ubuntu1 amd64 [installed]
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 runtime)

libwxgtk3.2-dev/plucky 3.2.6+dfsg-2ubuntu1 amd64
  wxWidgets Cross-platform C++ GUI toolkit (GTK 3 development)
```

This is just using vanilla Ubuntu Desktop 25.04, downloadable from here: https://ubuntu.com/download/desktop

guanyun-gudujian, 2025-05-29:
> vanilla Ubuntu Desktop 25.04

how about these commands:
```
sudo apt update
sudo apt install libwxgtk3.2-dev wx3.2-headers wx-common

```

d01, 2025-05-29:
Thank you!!

This is just Ubuntu with an NVidia GPU.

celstark, 2025-05-29:
> > `GDK_BACKEND=x11 ./Bambu_Studio_linux_fedora-v02.00.03.54.AppImage ` still gives the white screen (on Fedora here) and so does `GDK_BACKEND=x11 ./Bambu_Studio_ubuntu-24.04_PR-6688.AppImage `
> 
> can above gl test script run on your device?

Those same libraries don't exist on Fedora and I've not searched to find the Fedora / RPM libraries for wxWidgets

celstark, 2025-05-29:
I installed wxBase-devel to get wx-config, but ..
```
/dev/fd/63:2:10: fatal error: wx/wx.h: No such file or directory
compilation terminated.
cc1plus: fatal error: wxGTK-devel: No such file or directory
compilation terminated.
cc1plus: fatal error: isn't: No such file or directory
compilation terminated.
cc1plus: fatal error: installed: No such file or directory
compilation terminated.
cc1plus: fatal error: for: No such file or directory
compilation terminated.
cc1plus: fatal error: architecture: No such file or directory
compilation terminated.
cc1plus: fatal error: 'x86_64': No such file or directory
compilation terminated.
Test wayland backend
./glcanvastest: line 5: ./glcanvastest.exe: No such file or directory
Test x11 backend
./glcanvastest: line 5: ./glcanvastest.exe: No such file or directory
```

guanyun-gudujian, 2025-05-29:
seems that this is a wayland related issue.

we need to force use x11 （remove all wayland related stuff from BambuStudio & wxWidgets） or to support native wayland

guanyun-gudujian, 2025-05-29:
> OK, a few more installs (wxGTK-devel -- note, not wxGTK3-devel):
> 
> ## This program wasn't compiled with EGL support required under Wayland, either
> install EGL libraries and rebuild or run it under X11 backend by setting
> environment variable GDK_BACKEND=x11 before starting your program.
> ## ./src/unix/glx11.cpp(493): assert ""vi"" failed in wxGLContext(): invalid visual for OpenGL
> ## empty "Hello GL World" window
> Window you expect (on x11)

ok thank you very much

wayland only issue

celstark, 2025-05-29:
> seems that this is a wayland related issue.
> 
> we need to force use x11 （remove all wayland related stuff from BambuStudio & wxWidgets） or to support native wayland

Ayup -- given that even RHEL is going full Wayland now, much less all of GNOME, I'd go for door #2.

guanyun-gudujian, 2025-05-29:
> Ayup -- given that even RHEL is going full Wayland now, much less all of GNOME, I'd go for door [#2](https://github.com/bambulab/BambuStudio/pull/2).

yes, agreed
two steps to fix this issue:
phase 1. fallback to x11 for a workaround
phase 2. add wayland support

guanyun-gudujian, 2025-05-29:
@d01  @celstark   @keysking 
Hi  guys,
pls check if this package can work on your devices?  
thanks

guanyun-gudujian, 2025-05-29:
@d01   @celstark 
Hi guys,
pls help to check if this package can work on your device.
thanks
https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15319178759/artifacts/3219406837

celstark, 2025-05-30:
Initial load of a file showed a garbage / noise screen in the render area.  Get it to update by resizing the screen (not by rotating the model) and it cleans up and looks perfect.

[image]

guanyun-gudujian, 2025-05-30:
> Initial load of a file showed a garbage / noise screen in the render area. Get it to update by resizing the screen (not by rotating the model) and it cleans up and looks perfect.
> 
> [image]

ok thanks for your feedback

will take a look

if vsync enabled on your device?

matt3948, 2025-05-30:
> Initial load of a file showed a garbage / noise screen in the render area. Get it to update by resizing the screen (not by rotating the model) and it cleans up and looks perfect.
> 
> [image]

I'm experiencing the same with the AppImage, definitely promising!

On my system I do not believe vsync is enabled:

[image]

[image]

Thanks for your support on this, @guanyun-gudujian

guanyun-gudujian, 2025-06-04:
> Initial load of a file showed a garbage / noise screen in the render area. Get it to update by resizing the screen (not by rotating the model) and it cleans up and looks perfect.
> 
> [image]

@celstark 

pls try this one:
https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091

matt3948, 2025-06-04:
> [@matt3948](https://github.com/matt3948) pls help to check if this issue can be resolved by switching to x11 mode, thanks.
> 
> you can switch to x11 mode on the login screen
> 
> [image]

Yes, x11 does resolve the issue with the garbage / noise for _Bambu_Studio_ubuntu-22.04_V02.00.03.54.AppImage_

guanyun-gudujian, 2025-06-05:
> > [@matt3948](https://github.com/matt3948) pls help to check if this issue can be resolved by switching to x11 mode, thanks.
> > you can switch to x11 mode on the login screen
> > [image]
> 
> Yes, x11 does resolve the issue with the garbage / noise for _Bambu_Studio_ubuntu-22.04_V02.00.03.54.AppImage_ without having to resize the window.
> 
> [image]

ok

thanks for your feedback

celstark, 2025-06-08:
> [@celstark](https://github.com/celstark)
> 
> pls try this one: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091

Same issue - garbage on start, resize fixes it.

GuDuJian-J-Zhang, 2025-06-09:
> > [@celstark](https://github.com/celstark)
> > pls try this one: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091
> 
> Same issue - garbage on start, resize fixes it. Note, it also goes to garbage when I hit "Slice plate". That only happens on the first slice though. It's as if some array isn't getting cleared the first time it's used.

thanks, some customs announced that update nvidia driver can resolve this garbage rendering issue.

guanyun-gudujian, 2025-06-09:
> Same issue - garbage on start, resize fixes it. Note, it also goes to garbage when I hit "Slice plate". That only happens on the first slice though. It's as if some array isn't getting cleared the [...]

thanks, some customs announced that update nvidia driver can resolve this garbage rendering issue: 
https://github.com/bambulab/BambuStudio/issues/5753#issuecomment-2954168042

jeremie-lesage, 2025-06-09:
I try Bambu_Studio_ubuntu-22.04_V02.00.03.54.AppImage on fedora 42, it works fine. (GeForce RTX 2060, Driver Version: 575.57.08, CUDA Version: 12.9)

guanyun-gudujian, 2025-06-11:
> Having the same issue on wayland with latest Nvidia drivers on Arch Linux
> 
> > pls try this one: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091
> 
> This version is the only one working. Flatpak and AUR versions only runs under x11

Thanks for your feedback
Also have the tearing rendering issue right?

jeremie-lesage, 2025-06-21:
It's working for me. thanks

EtheriousNight, 2025-07-05:
It would be wonderful if this gets merged into this official repo. I had the same Issue, and it's now working with your fix ^^

guanyun-gudujian, 2025-07-05:
> It would be wonderful if this gets merged into this official repo. I had the same Issue on fedora 42 with wayland + nvidia, and it's now working with your fix ^^

The official package with this fix will be released soon

EtheriousNight, 2025-07-18:
Hi, I installed the AppImage Version 2.2.0.60, and it seems like it's not working. I also needed to install additional Packages compared to your fixed Version. 

The Packages are webkit2gtk3, libsoup

GuDuJian-J-Zhang, 2025-07-18:
> Hi, I installed the AppImage Version 2.2.0.60, and it seems like it's not working. I also needed to install additional Packages compared to your fixed Version.
> 
> The Packages are webkit2gtk3, libsoup

it's maybe a packaging issue.

can it rendering 3d view after installing above libs？

guanyun-gudujian, 2025-07-18:
> Hi, I installed the AppImage Version 2.2.0.60, and it seems like it's not working. I also needed to install additional Packages compared to your fixed Version.
> 
> The Packages are webkit2gtk3, libsoup

it's maybe a packaging issue.
can it rendering 3d view after installing above libs？

EtheriousNight, 2025-07-18:
Unfortunately, no. :(

guanyun-gudujian, 2025-07-18:
> Unfortunately, no. :(

still black screen?

EtheriousNight, 2025-07-18:
> > Unfortunately, no. :(
> 
> still black screen?

[image]

Yes.

guanyun-gudujian, 2025-07-18:
> > > Unfortunately, no. :(
> > 
> > 
> > still black screen?
> 
> <img alt="Image" width="1722" height="1388" src="https://private-user-images.githubusercontent.com/56686174/467940732-30661e3d-cc93-4fc0-b678-f6f0849fa9eb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI4MjgyNTgsIm5iZiI6MTc1MjgyNzk1OCwicGF0aCI6Ii81NjY4NjE3NC80Njc5NDA3MzItMzA2NjFlM2QtY2M5My00ZmMwLWI2NzgtZjZmMDg0OWZhOWViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE4VDA4MzkxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdlZjc1ZjViM2ZiYjJjN2U5Mzk1YjhmOTJmMDBkMTI2ZTNkOTliMmFiMDI4MjNiZThiZTVjNTlhYWVjYmIxY2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ZBzQepUPJExfwR1WuG0GpWIMNHtPjhre3MPszvUDp1c">
> Yes.

maybe a packing issue , so sorry for that

i'll see

guanyun-gudujian, 2025-07-18:
> Unfortunately, no. :(

pls help to check which one work, thanks

version1 : (without platform checking)
for ubuntu 24.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366488248/artifacts/3562148808
for ubuntu 22.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366488248/artifacts/3562130141

version2: (with platform checking)
for ubuntu 24.04:  https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366503233/artifacts/3562161529
for ubuntu 22.04:  https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366503233/artifacts/3562154867

PPPDUD, 2025-08-01:
I am also running Ubuntu 25.04 on Wayland with a 3060, have you all found a solution yet?

EtheriousNight, 2025-08-03:
> [@celstark](https://github.com/celstark)
>
> pls try this one: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091

@guanyun-gudujian I downloaded the AppImage from this Artifact Link. Could you do it the same way with the test versions? 

@PPPDUD You can use the AppImage from the above message. It works on Fedora 42 with NVIDIA Drivers + Wayland, and I'm using it with the software "GearLever". Hope it works on Ubuntu ^^

EtheriousNight, 2025-08-04:
@guanyun-gudujian It seems like both versions work on fedora 42 with nvidia and wayland. 🤔

guanyun-gudujian, 2025-08-05:
> [@guanyun-gudujian](https://github.com/guanyun-gudujian) It seems like both versions work on fedora 42 with nvidia and wayland. 🤔

thanks

how about this one:

ubuntu22.04: https://github.com/bambulab/BambuStudio/actions/runs/16650306777/artifacts/3659171461
ubuntu24.04: https://github.com/bambulab/BambuStudio/actions/runs/16650306777/artifacts/3659182163

EtheriousNight, 2025-08-05:
@guanyun-gudujian Worked 👍

PPPDUD, 2025-08-05:
> > > Initial load of a file showed a garbage / noise screen in the render area. Get it to update by resizing the screen (not by rotating the model) and it cleans up and looks perfect.
> > > [image]
> > 
> > 
> > [@celstark](https://github.com/celstark)
> > pls try this one: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091
> 
> [@guanyun-gudujian](https://github.com/guanyun-gudujian) I downloaded the AppImage from this Artifact Link. Could you do it the same way with the test versions from ur last message? 😊
> 
> [@PPPDUD](https://github.com/PPPDUD) You can use the AppImage from the above message. It works on Fedora 42 with NVIDIA Drivers + Wayland, and I'm using it with the software "GearLever". Hope it works on Ubuntu ^^

When I ran that executable, it produced the following error: `/tmp/.mount_Bambu_ZIOcjy/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`

guanyun-gudujian, 2025-08-06:
> When I ran that executable, it produced the following error: `/tmp/.mount_Bambu_ZIOcjy/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object [...]

manually install it
thanks

guanyun-gudujian, 2025-08-06:
> [@guanyun-gudujian](https://github.com/guanyun-gudujian) Worked 👍

that is so crazy
have you try the latest offical package?
the codes are same

PPPDUD, 2025-08-06:
> manually install it thanks

I'm new to Ubuntu, how am I supposed to install AppImages? Do you mean using one of those programs that adds the AppImage to my program selector?

guanyun-gudujian, 2025-08-07:
> > > [@guanyun-gudujian](https://github.com/guanyun-gudujian) Worked 👍
> > 
> > 
> > that is so crazy have you try the latest offical package? the codes are same
> > here pls: https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.68
> 
> Hi [@guanyun-gudujian](https://github.com/guanyun-gudujian) this Version works too. In Version v02.02.00.60 there was a Fedora AppImage that I used for testing and this specific Version of the AppImage was the problematic one... The Ubuntu AppImage builds work in both Versions (v02.02.00.60 and v02.02.00.68) work fine an.
> 
> v02.02.00.68 does not even build a specific fedora AppImage xD
> 
> Sorry for the trouble 😅

ok happy to here that. congrats